esphome/wake-word-voice-assistants #11: micro-wake-word improvements

mateuszdrab, 2024-08-12:
Hello

I've recently adopted the wake word variant of the firmware for my m5stack atom echo and noticed an amazing improvement in terms of detection reliability.

I'd like to provide some feedback/request a feature:

- old configuration had the option to disable wake word detection with a switch, can this be added back? It's great for preventing detections when I enable house wide DnD.
- is the ESP Muse Luxe compatible? I'd love micro-wake-word on it as well.

Thanks!

mateuszdrab, 2024-08-13:
Working on those improvements in my [fork](https://github.com/mateuszdrab/wake-word-voice-assistants)

I've been able to restore the use wake word switch and port the changes to the ESP Muse Luxe.

Will PR the changes soon after some more testing.

chpego, 2024-11-08:
Hi @mateuszdrab ,
Any update on this ? after a few weeks, is it stable now? would it be possible to change this request?

mateuszdrab, 2024-11-08:
> Hi @mateuszdrab ,
> Any update on this ? after a few weeks, is it stable now? would it be possible to change this request?

I have not updated the PR because I'm having issues with my devices

I think wake word detection causes lots of load on the CPU because once a request is made, the response is very stuttery and makes the devices unusable.

Perhaps it can be worked around by restarting the wake word detection with a delay, I'll need to look at this again as I've not had much time

mateuszdrab, 2024-12-01:
Hi @chpego 

I've had a look at what happens, and generally the issue is that the first wake word detection, request and response work fine; however, after that, subsequent request result in an error loop with messages similar to the below:

```
[22:46:57][D][i2s_audio.speaker:108]: Starting Speaker
[22:46:57][D][i2s_audio.speaker:126]: Stopped Speaker
[22:46:57][W][i2s_audio.speaker:148]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:46:57][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
```

I think I'm not properly stopping/starting something and messing up the internal state of things.

If anyone more experienced could deploy the code from my fork, I'd appreciate some feedback.

<details>

<summary>Full log from the repro attempt</summary>

```
[22:46:30][D][switch:012]: 'Use wake word' Turning ON.
[22:46:30][D][switch:055]: 'Use wake word': Sending state ON
[22:46:30][D][ring_buffer:034]: Created ring buffer with size 2048
[22:46:30][D][micro_wake_word:399]: Resetting buffers and probabilities
[22:46:30][D][micro_wake_word:195]: State changed from IDLE to START_MICROPHONE
[22:46:30][D][light:036]: 'M5stack Atom Echo 804674' Setting:
[22:46:30][D][light:047]:   State: ON
[22:46:30][D][light:051]:   Brightness: 60%
[22:46:30][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[22:46:30][W][component:237]: Component api took a long time for an operation (84 ms).
[22:46:30][W][component:238]: Components should block for at most 30 ms.
[22:46:30][D][micro_wake_word:107]: Starting Microphone
[22:46:30][D][micro_wake_word:195]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[22:46:30][D][esp-idf:000]: I (1069302) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[22:46:30][D][micro_wake_word:195]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[22:46:31][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:46:31][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:46:33][D][micro_wake_word:162]: The 'Okay Nabu' model sliding average probability is 0.984 and most recent probability is 1.000
[22:46:33][D][micro_wake_word:123]: Wake Word 'Okay Nabu' Detected
[22:46:33][D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[22:46:33][D][micro_wake_word:129]: Stopping Microphone
[22:46:33][D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[22:46:33][D][esp-idf:000]: I (1072928) I2S: DMA queue destroyed

[22:46:33][D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE
[22:46:33][D][voice_assistant:512]: State changed from IDLE to START_MICROPHONE
[22:46:33][D][voice_assistant:518]: Desired state set to START_PIPELINE
[22:46:33][D][voice_assistant:222]: Starting Microphone
[22:46:33][D][ring_buffer:034]: Created ring buffer with size 16384
[22:46:33][D][voice_assistant:512]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[22:46:33][D][esp-idf:000]: I (1072978) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[22:46:33][D][voice_assistant:512]: State changed from STARTING_MICROPHONE to START_PIPELINE
[22:46:33][D][voice_assistant:276]: Requesting start...
[22:46:33][D][voice_assistant:512]: State changed from START_PIPELINE to STARTING_PIPELINE
[22:46:33][D][voice_assistant:533]: Client started, streaming microphone
[22:46:33][D][voice_assistant:512]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[22:46:33][D][voice_assistant:518]: Desired state set to STREAMING_MICROPHONE
[22:46:33][D][voice_assistant:635]: Event Type: 1
[22:46:33][D][voice_assistant:638]: Assist Pipeline running
[22:46:34][D][voice_assistant:635]: Event Type: 3
[22:46:34][D][voice_assistant:649]: STT started
[22:46:34][D][light:036]: 'M5stack Atom Echo 804674' Setting:
[22:46:34][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[22:46:34][D][light:109]:   Effect: 'Slow Pulse'
[22:46:35][D][voice_assistant:635]: Event Type: 11
[22:46:35][D][voice_assistant:795]: Starting STT by VAD
[22:46:36][D][voice_assistant:635]: Event Type: 12
[22:46:36][D][voice_assistant:799]: STT by VAD end
[... 187 lines cut ...]
[22:46:58][D][i2s_audio.speaker:108]: Starting Speaker
[22:46:58][D][i2s_audio.speaker:126]: Stopped Speaker
[22:46:58][W][i2s_audio.speaker:148]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:46:58][E][voice_assistant:819]: Cannot receive audio, buffer is full
[22:46:58][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
[22:46:58][W][i2s_audio.speaker:148]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:46:58][D][i2s_audio.speaker:126]: Stopped Speaker
[22:46:58][W][i2s_audio.speaker:148]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:46:58][E][voice_assistant:819]: Cannot receive audio, buffer is full
[22:46:58][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
[22:46:58][D][i2s_audio.speaker:126]: Stopped Speaker
[22:46:58][W][i2s_audio.speaker:148]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:46:58][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
[22:46:58][D][i2s_audio.speaker:108]: Starting Speaker
[22:46:58][D][i2s_audio.speaker:126]: Stopped Speaker
[22:46:58][W][i2s_audio.speaker:148]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:46:58][E][voice_assistant:819]: Cannot receive audio, buffer is full
[22:46:58][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
[22:46:58][E][voice_assistant:819]: Cannot receive audio, buffer is full
[22:46:58][D][i2s_audio.speaker:126]: Stopped Speaker
[22:46:58][W][i2s_audio.speaker:148]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:46:58][E][voice_assistant:819]: Cannot receive audio, buffer is full
```

</details>

s00500, 2025-02-19:
running into the same issue here...

samperk1, 2025-02-25:
[21:09:57][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE
[21:09:57][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
[21:09:57][D][i2s_audio.speaker:111]: Starting Speaker
[21:09:57][D][i2s_audio.speaker:129]: Stopped Speaker
 Hi has anyone figured out whats going on with this? I'm having the same problem.

s00500, 2025-02-25:
For the invalid state issue: the problem is if you have a mic that is still listening while trying to write to the speaker, making sure you stop capturing for micro wake word before playing was the fix for me

Some logic like that has also been added to the newer M5Stack configs I think

closed 2025-04-30 by jesserockz