esphome/wake-word-voice-assistants #14: [FEATURE REQUREST] Support for Radar-Sensor of the S3-Box3 "Sensor-Dock"

treysullivent, 2024-09-03:
The Espressif S3-Box3 comes with a ESP32-S3-BOX-3-SENSOR dock that includes an IR transmitter/receiver, a temperature and humity sensor, and a mm wave presence detection sensor. 

It would be really nice to expose these sensors to Home Assistant.

igorsantos07, 2024-10-21:
I would suggest editing the issue title to reflect you're talking about all the sensors, not only the radar-sensor.

Other than that, I should mention it was actually quite frustrating to realize all those sensors go to dead when you install the firmware. Sadly, no tutorial or docs explain that - not even the "find the device for buying in this link" sections. I could have saved a couple bucks (or many, if you consider exchange rate and taxes) if someone had written an extra sentence on the docs 🤦‍♂️

closed 2025-05-08 by github-actions[bot]

igorsantos07, 2025-05-09:
1. nostalebots.xyz
2. I solved this issue with [Bigbobba's awesome repo](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome) which, contrary to many of the HA's repos, is not abandoned lol