esphome/wake-word-voice-assistants #18: ESP32-S3-Box3 ESPHome V2024.9.0 Update-Problem

7542, 2024-09-18:
When i try to Update My ESP32-S3-Box3 i get:

```
INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/esp32-s3-box-3-2d99b4.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

display.ili9xxx: [source /data/packages/dec89df9/esp32-s3-box-3/esp32-s3-box-3.yaml:688]
  
  'invert_colors' is a required option for [display.ili9xxx].
  platform: ili9xxx
  id: s3_box_lcd
  model: S3BOX
  data_rate: 40MHz
  cs_pin: 5
  dc_pin: 4
  reset_pin: 
    number: 48
    inverted: True
  update_interval: never
  pages: 
    - id: idle_page
```

dj-sf, 2024-09-18:
having this same issue.  Proximate cause seems to be that there is a required parameter missing from [this yaml file](https://github.com/esphome/wake-word-voice-assistants/blob/79ec1c60f4f36ae18b2a5a3765c02e1ca2a4bfab/esp32-s3-box-3/esp32-s3-box-3.yaml#L688).  

Maybe due to some patch being applied automatically.

Investigating now.

dj-sf, 2024-09-18:
I opened another issue for this earlier: https://github.com/esphome/issues/issues/6257

dj-sf, 2024-09-18:
PR up with a fix: https://github.com/esphome/wake-word-voice-assistants/pull/19

PR #19 (2024-09-19): Add missing invert_colors yaml prop
Confirmed that this works by copying the updated yaml file and manually installing it on my local Esp32 S3 Box 3

Closes #18
closed 2024-09-19 by jesserockz