esphome/wake-word-voice-assistants #26: binary_sensor → assist_satellite (deprecation)

only1thor, 2024-10-04:
according to the repair notification i home assistant 2024.10, the `binary_sensor.m5stack_atom_echo_8a1d90_assist_in_progress` is deprecated and replaced by `assist_satellite`. 

I belive the relevant section is [here](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml#L159). 

exact text from the repair notification is:

> Repair issue
Warning ⸱ Reported by ESPHome
⚠ This stops working in version 2025.4. Please address before upgrading.
>
> The ESPHome in progress binary sensor `binary_sensor.m5stack_atom_echo_8a1d90_assist_in_progress` is deprecated.
>
> Migrate your configuration to use the corresponding `assist_satellite` entity and then click SUBMIT to disable the in progress binary sensor and fix this issue.

synesthesiam, 2024-10-06:
Please use the state of the Assist satellite entity instead going forward.

ijc, 2024-10-06:
I have this same message (except `binary_sensor.esp32_s3_box_3_«...»_assist_in_progress`).

> I belive the relevant section is [here](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml#L159).

I found this message confusing for a while too. I don't think it is related to this binary sensor.

I think what we need to do (and what I have done) is check that we have no automations etc referencing `binary_sensor.*_assist_in_progress` and then it is safe to submit to disable the entry. The sensor will disappear eventually in 2025.4.

IOW I think the message is telling us to stop *consuming* the entity, not that we need to do something to stop our esphome devices from *providing* it. I think it is/was actually provided by the wyoming component within HA.

The PR which added this warning was https://github.com/home-assistant/core/pull/125805 which led me to https://github.com/home-assistant/core/pull/125806 which is the one which actually disabled the sensor.

closed 2024-11-22 by jesserockz