esphome/wake-word-voice-assistants #30: ESPHome update 2024.10.1 broke local wake word pipeline

crissmil, 2024-10-23:
after updating home assistant ESPhome to 2024.10.1 and Atom Echo satellites, wake word detection is not working anymore.
Upon detection of wake-word, pipeline is changing state: "State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE" and assistant is not working. I have 3 atom echo satellites, all of them with same issue.

please see below log:
=========
INFO ESPHome 2024.10.1
INFO Reading configuration /config/esphome/m5stack-atom-echo-23bcf8.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Starting log output from 10.0.216.67 using esphome API
INFO Successfully connected to m5stack-atom-echo-23bcf8 @ 10.0.216.67 in 0.065s
INFO Successful handshake with m5stack-atom-echo-23bcf8-23bcf8 @ 10.0.216.67 in 0.483s
[19:33:04][I][app:100]: ESPHome version 2024.10.1 compiled on Oct 22 2024, 20:07:01
[19:33:04][C][wifi:600]: WiFi:
[19:33:04][C][wifi:428]:   Local MAC: E8:6B:EA:23:BC:F8
[19:33:04][C][wifi:433]:   SSID: [redacted]
[19:33:04][C][wifi:436]:   IP Address: 10.0.216.67
[19:33:04][C][wifi:440]:   BSSID: [redacted]
[19:33:04][C][wifi:441]:   Hostname: 'm5stack-atom-echo-23bcf8-23bcf8'
[19:33:04][C][wifi:443]:   Signal strength: -56 dB ▂▄▆█
[19:33:04][C][wifi:447]:   Channel: 6
[19:33:04][C][wifi:448]:   Subnet: 255.255.255.0
[19:33:04][C][wifi:449]:   Gateway: 10.0.216.1
[19:33:04][C][wifi:450]:   DNS1: 0.0.0.0
[19:33:04][C][wifi:451]:   DNS2: 0.0.0.0
[19:33:04][C][logger:185]: Logger:
[19:33:04][C][logger:186]:   Level: DEBUG
[19:33:04][C][logger:188]:   Log Baud Rate: 115200
[19:33:04][C][logger:189]:   Hardware UART: UART0
[19:33:04][C][esp32_rmt_led_strip:187]: ESP32 RMT LED Strip:
[19:33:04][C][esp32_rmt_led_strip:188]:   Pin: 27
[19:33:04][C][esp32_rmt_led_strip:189]:   Channel: 0
[19:33:04][C][esp32_rmt_led_strip:214]:   RGB Order: GRB
[19:33:04][C][esp32_rmt_led_strip:215]:   Max refresh rate: 0
[19:33:04][C][esp32_rmt_led_strip:216]:   Number of LEDs: 1
[19:33:04][C][template.select:065]: Template Select 'Wake word engine location'
[19:33:04][C][template.select:066]:   Update Interval: 60.0s
[19:33:04][C][template.select:069]:   Optimistic: YES
[19:33:04][C][template.select:070]:   Initial Option: On device
[19:33:04][C][template.select:071]:   Restore Value: YES
[19:33:04][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Button'
[19:33:04][C][gpio.binary_sensor:016]:   Pin: GPIO39
[19:33:04][C][light:103]: Light 'M5Stack Atom Bathroom 23bcf8'
[19:33:04][C][light:105]:   Default Transition Length: 0.0s
[19:33:04][C][light:106]:   Gamma Correct: 2.80
[19:33:04][C][template.switch:068]: Template Switch 'Use listen light'
[19:33:04][C][template.switch:091]:   Restore Mode: disabled 
[19:33:04][C][template.switch:057]:   Optimistic: YES
[19:33:05][C][template.switch:068]: Template Switch 'timer_ringing'
[19:33:05][C][template.switch:091]:   Restore Mode: always OFF
[19:33:05][C][template.switch:057]:   Optimistic: YES
[19:33:05][C][template.switch:068]: Template Switch 'Listen Once'
[19:33:05][C][template.switch:091]:   Restore Mode: always OFF
[19:33:05][C][template.switch:057]:   Optimistic: YES
[19:33:05][C][template.switch:068]: Template Switch 'Use this satellite'
[19:33:05][C][template.switch:091]:   Restore Mode: disabled 
[19:33:05][C][template.switch:057]:   Optimistic: YES
[19:33:05][C][factory_reset.button:011]: Factory Reset Button 'Factory reset'
[19:33:05][C][factory_reset.button:011]:   Icon: 'mdi:restart-alert'
[19:33:05][C][captive_portal:089]: Captive Portal:
[19:33:05][C][mdns:116]: mDNS:
[19:33:05][C][mdns:117]:   Hostname: m5stack-atom-echo-23bcf8-23bcf8
[19:33:05][C][esphome.ota:073]: Over-The-Air updates:
[19:33:05][C][esphome.ota:074]:   Address: 10.0.216.67:3232
[19:33:05][C][esphome.ota:075]:   Version: 2
[19:33:05][C][safe_mode:018]: Safe Mode:
[19:33:05][C][safe_mode:020]:   Boot considered successful after 60 seconds
[19:33:05][C][safe_mode:021]:   Invoke after 10 boot attempts
[19:33:05][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[19:33:05][C][api:140]: API Server:
[19:33:05][C][api:141]:   Address: 10.0.216.67:6053
[19:33:05][C][api:143]:   Using noise encryption: YES
[19:33:05][C][micro_wake_word:051]: microWakeWord:
[19:33:05][C][micro_wake_word:052]:   models:
[19:33:05][C][micro_wake_word:015]:     - Wake Word: Hey Jarvis
[19:33:05][C][micro_wake_word:016]:       Probability cutoff: 0.970
[19:33:05][C][micro_wake_word:017]:       Sliding window size: 5
[19:33:05][C][micro_wake_word:015]:     - Wake Word: Alexa
[19:33:05][C][micro_wake_word:016]:       Probability cutoff: 0.900
[19:33:05][C][micro_wake_word:017]:       Sliding window size: 5
[19:33:05][C][micro_wake_word:021]:     - VAD Model
[19:33:05][C][micro_wake_word:022]:       Probability cutoff: 0.500
[19:33:05][C][micro_wake_word:023]:       Sliding window size: 5
[19:33:19][D][micro_wake_word:162]: The 'Alexa' model sliding average probability is 0.990 and most recent probability is 1.000
[19:33:19][D][micro_wake_word:123]: Wake Word 'Alexa' Detected
[19:33:19][D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[19:33:19][D][micro_wake_word:129]: Stopping Microphone
[19:33:19][D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[19:33:19][D][esp-idf:000]: I (381980) I2S: DMA queue destroyed

[19:33:19][D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE
[19:33:19][D][voice_assistant:510]: State changed from IDLE to START_MICROPHONE
[19:33:19][D][voice_assistant:516]: Desired state set to START_PIPELINE
WARNING m5stack-atom-echo-23bcf8-23bcf8 @ 10.0.216.67: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for m5stack-atom-echo-23bcf8-23bcf8 @ 10.0.216.67
WARNING Disconnected from API
INFO Successfully connected to m5stack-atom-echo-23bcf8-23bcf8 @ 10.0.216.67 in 0.084s
INFO Successful handshake with m5stack-atom-echo-23bcf8-23bcf8 @ 10.0.216.67 in 0.156s
========

Pat0856, 2024-10-23:
I have exactly the same issue with my m5stack-atom-echo under Esphome 2024.10.1. 
The most annoying is that the device is over and over disconnecting  from wifi and it's no more possible 
to restore to esphome 2024.10.0 by wifi. Fortunately I could restore to Esphome 2024.10.0 
by USB-connection and t's now successfully working again

crissmil, 2024-10-24:
I confirm on 2024.10.2 is working again

closed 2024-11-22 by jesserockz