esphome/wake-word-voice-assistants #33: Error when adopting the ESP32-S3-BOX-3B into Esphome

alemuro, 2024-11-20:
Hello, 

I've been trying to adopt my new ESP32-S3-BOX-3B into  Esphome, but when I click on Install, I face the following error. I think this happened due to a rate limit on Github, and since then I alway face this error.

I tried removing folders inside the container with no luck. 

[image]

Please, would it be possible to receive some guidance/hint on what should I do to "cleanup" the Esphome config (without losing all data)? And, did you see this error before? I didn't found anything on the Internet 😢 

Many thanks in advance! 🙏

ericvb, 2024-11-20:
I've no solution, but asked the same question here: https://community.home-assistant.io/t/compiling-esp32-s3-box-3-fails-and-gives-failed-config-font-is-missing-791-glyphs/797536

hugobloem, 2024-11-21:
This is fixed in #34

MikeDembek, 2024-11-21:
I'm seeing this following #34 - here's my yaml:

```
substitutions:
  name: esp32-s3-box-3
  friendly_name: somename
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: somekey

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

CamronBorealis, 2024-11-21:
I am still seeing this as well

hugobloem, 2024-11-21:
It seems like it caches the file locally somewhere. I worked around this by removing the add-on from Home Assistant and re-installing it, but this isn't an elegant solution.

MikeDembek, 2024-11-21:
> It seems like it caches the file locally somewhere. I worked around this by removing the add-on from Home Assistant and re-installing it, but this isn't an elegant solution.

It seemed like this to me too.  I did a clean build files but it didn't appear to make any difference.

jesserockz, 2024-11-21:
The package is cached for 24 hours from the last successful grab from github. 

You can change:
```yaml
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
```

to 

```yaml
packages:
  esphome.voice-assistant: 
    url: https://github.com/esphome/wake-word-voice-assistants
    file: esp32-s3-box-3/esp32-s3-box-3.yaml
    ref: main
    refresh: 0s
```

and it should get the new file right away

alemuro, 2024-11-21:
Many thanks!!! Seems it is fixed 😄 

I have a [different error now](https://github.com/esphome/issues/issues/6460), so there is some advance.

I guess this thread can be closed then.

closed 2024-11-21 by alemuro