esphome/wake-word-voice-assistants #36: Regression on displaying non-ASCII characters

bors-ltd, 2024-11-22:
Following https://github.com/esphome/wake-word-voice-assistants/pull/34 (I can only guess), my ESP32-S3-Box-3 now displays black rectangles instead of accentuated letters.

Here are some (literal) screenshots:
[image]
[image]

I don't really have a solution, as the firmware won't compile without that commit. Replacing the font altogether I guess.

It's not breaking, the voice assistant is still functional, but a slight regression nonetheless.

HLandstrom, 2024-11-22:
I have the same issue.
Took a picture too just in case.

The bars are supposed to be å, ä and ö.

[image]

drogfild, 2024-12-08:
After some debugging, I’ve pinpointed the issue to the configuration of font glyph sets in ESPHome. Currently, when using Google Fonts (gfonts) in YAML configuration, users cannot specify glyph sets to tailor the included character range for rendering. This limitation leads to missing characters (e.g., extended Latin characters like ä, ö, å) even when the font itself supports them.

Here’s an example configuration for context:
```
font:
  - file:
      type: gfonts
      family: Figtree
      weight: 300
      italic: true
    id: font_request
    size: 15
    glyphsets:
      - GF_Latin_Core
  - file:
      type: gfonts
      family: Figtree
      weight: 300
    id: font_response
    size: 15
    glyphsets:
      - GF_Latin_Core
```

By adding glyphsets, the user could define a predefined character set (such as GF_Latin_Core or other supported sets) to ensure the correct glyphs are loaded for their use case. This flexibility would improve support for multilingual text rendering, especially for non-ASCII characters.

https://github.com/googlefonts/glyphsets/blob/main/GLYPHSETS.md#gf-latin-core
https://esphome.io/components/font.html#font-renderer-component

Request:
It would be fantastic if someone could implement this feature as a pull request. It would significantly enhance Assistanses ESPHome's text rendering capabilities and enable more robust font handling for various languages.

Looking forward to feedback or collaboration on this! Let me know if more details or testing are required.

drogfild, 2024-12-10:
I made the PR and it's now waiting for a review.

HLandstrom, 2024-12-11:
Thank you!
I tried your PR on mine and it works as it should now

olskar, 2025-03-20:
Is this merged yet? Still have the issue

closed 2025-04-30 by jesserockz

drogfild, 2025-04-30:
Why set Not planned? Fix in #41  fixed this issue as far as I know.