esphome/wake-word-voice-assistants #42: Projects using ESP-IDF don't build with the 2024.12 betas

rccoleman, 2024-12-16:
### The problem

None of my Assist projects build with the ESPHome 2024.12 betas.  They all fail with a build error in ESP-IDF related to the JSMN component.

### Which version of ESPHome has the issue?

2024.12.0b0/b1/b2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.12.0b3

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-Box-3

### Component causing the issue

jsnm

### YAML Config

```yaml
Top-level config:

substitutions:
  name: fr-esp32-s3
  friendly_name: Family Room ESP32-S3

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

packages:
  esp32-s3-box-3: !include wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml

# Enable logging
logger:

# Enable Home Assistant API
api:

audio:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:

captive_portal:

button:
  - platform: restart
    name: "Restart"

Using the stock MWW ESP32-S3-Box-3 config: https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box-3/esp32-s3-box-3.yaml
```
[... 305 lines cut ...]
Installing collected packages: wheel, urllib3, typing-extensions, tqdm, pyyaml, python-dotenv, pyparsing, pycparser, kconfiglib, jsonref, idna, future, colorama, click, charset-normalizer, certifi, annotated-types, requests, pydantic-core, esp-idf-kconfig, cffi, requests-toolbelt, requests-file, pydantic, cryptography, pydantic-settings, idf-component-manager
Successfully installed annotated-types-0.7.0 certifi-2024.12.14 cffi-1.17.1 charset-normalizer-3.4.0 click-8.1.7 colorama-0.4.6 cryptography-41.0.7 esp-idf-kconfig-1.5.0 future-1.0.0 idf-component-manager-2.0.4 idna-3.10 jsonref-1.1.0 kconfiglib-14.1.0 pycparser-2.22 pydantic-2.10.3 pydantic-core-2.27.1 pydantic-settings-2.7.0 pyparsing-3.2.0 python-dotenv-1.0.1 pyyaml-6.0.2 requests-2.32.3 requests-file-2.1.0 requests-toolbelt-1.0.0 tqdm-4.67.1 typing-extensions-4.12.2 urllib3-1.26.20 wheel-0.45.1
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/home/rcoleman/esphome/.esphome/build/fr-esp32-s3/.pioenvs/fr-esp32-s3/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /home/rcoleman/.platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /home/rcoleman/.platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /home/rcoleman/.platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /home/rcoleman/.platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

================================================ [FAILED] Took 86.56 seconds ===============================================
```

### Additional information

If I install ESPHome 2024.11.3, the same projects build and install just fine.  I get the same failure if I just try to build exactly the released config from https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box-3/esp32-s3-box-3.yaml as is.

jesserockz, 2024-12-16:
To use the s3 boxes with ESPHome 2024.12 or above will require some reworking of the audio drivers due to the craziness of esp-adf and esp-idf. Work has begun on this to remove the esp-adf dependency but is not complete yet. There is no timeline available.

Jesse

rccoleman, 2024-12-17:
I see the same behavior with the M5Stack Atom Echo clean build (not an ESP32-S3).  Is that expected?  My Onju devices fail the same way, but that's another ESP32-S3.  Basically, all of my assist devices fail in the same way now with the beta.

```
rcoleman@rcoleman-linux ...docker/esphome/config % ./install office-assist.yaml     
INFO ESPHome 2024.12.0b2
INFO Reading configuration office-assist.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Core config, version or integrations changed, cleaning build files...
INFO Cloning https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Fetching mdns-v1.3.2
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing office-assist (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
---------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/pioarduino/platform-espressif32.git#51.03.07
INFO Installing git+https://github.com/pioarduino/platform-espressif32.git#51.03.07
git version 2.39.5
Cloning into '/root/.platformio/.cache/tmp/pkg-installing-ivunrjin'...
remote: Enumerating objects: 914, done.
remote: Counting objects: 100% (914/914), done.
remote: Compressing objects: 100% (719/719), done.
remote: Total 914 (delta 315), reused 702 (delta 161), pack-reused 0 (from 0)
Receiving objects: 100% (914/914), 1.11 MiB | 4.69 MiB/s, done.
Resolving deltas: 100% (315/315), done.
Note: switching to 'ec17486ffa451374e108b72be07e9c8fa83338e8'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Platform Manager: espressif32@51.3.6+sha.ec17486 has been installed!
INFO espressif32@51.3.6+sha.ec17486 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
INFO Installing https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf@3.50105.0 has been installed!
INFO framework-espidf@3.50105.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 12.2.0+20230208
INFO Installing espressif/toolchain-xtensa-esp32 @ 12.2.0+20230208
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@12.2.0+20230208 has been installed!
INFO toolchain-xtensa-esp32@12.2.0+20230208 has been installed!
Tool Manager: Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
INFO Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-esp32ulp@2.35.0-20220830 has been installed!
INFO toolchain-esp32ulp@2.35.0-20220830 has been installed!
Tool Manager: Installing espressif/tool-xtensa-esp-elf-gdb @ ~12.1.0
INFO Installing espressif/tool-xtensa-esp-elf-gdb @ ~12.1.0
Downloading  [####################################]  100%
[... 143 lines cut ...]
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
.....NOTICE: Updating lock file at /config/.esphome/build/office-assist/dependencies.lock
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/config/.esphome/build/office-assist/.pioenvs/office-assist/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /root/.platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /root/.platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /root/.platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /root/.platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

============================================================== [FAILED] Took 82.34 seconds ==============================================================
```

BigBobbas, 2024-12-17:
defining 
```yaml
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```
    will compile on the S3-Box-3 Project firmware - not tested on Echo but assume it's the same deal

sla004, 2024-12-18:
> ```
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```
> 
> That fixed it for me on the Atom echo. Thanks!

Where did you set this?

BigBobbas, 2024-12-18:
if using the adopted firmware from a project install and don't have the `esp32:` block, you can add the following
```yaml
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

Mariusthvdb, 2024-12-18:
thanks BB, for now I have removed and reinstalled the Add-on, as I was instructed that should clean the build files (and probably downloads them again?)

I had been relocated to https://github.com/esphome/firmware/tree/main/voice-assistant/ under the link to GitHub, so that seems to be not the same as your link above.

all in all this seems way more involved, that simply hitting the update button, I hope this process will get some love. Not a tinkerer myself at all, not a complete noob either..

eamonnsullivan, 2024-12-19:
Pinning the idf version, as suggested in [this comment](https://github.com/esphome/wake-word-voice-assistants/issues/42#issuecomment-2550985988),  _does_ work for me, but I then get a warning in the logs. 

```
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
```

Mariusthvdb, 2024-12-19:
fwiw, I didnt manage to update the Atom Echo and ignored the repair in Home Assistant for now.
hope there will be a fix for this in the future updates

Mis1ify, 2024-12-21:
Same problem on ESP-S3-BOX-3, i can't upgrade the firmware, even with the newest 2024.12.2 stable version of the add-on. I obviously cleaned the build files.

Clasper101, 2024-12-21:
This:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
worked for my Atom Echo's but not my S3Box3.
I get this error:
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/voice-lounge/.pioenvs/voice-lounge/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

This applies to both 2024.12.1 and 2024.12.2 updates

Clasper101, 2024-12-21:
I've just re-tried to install on my S3Box3 and got an error again, although a different one, it got a lot further into the compiling before it failed. The log is below.
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/voice-lounge.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing voice-lounge (board: esp32s3box; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ESPMicroSpeechFeatures @ 1.1.0
Generating assembly for .pioenvs/voice-lounge/duer_profile.S
Compiling .pioenvs/voice-lounge/src/esphome/components/api/api_connection.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/api_server.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/list_entities.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/proto.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/voice-lounge/src/esphome/components/api/user_services.o
Compiling .pioenvs/voice-lounge/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/voice-lounge/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/voice-lounge/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/voice-lounge/src/esphome/components/button/button.o
Compiling .pioenvs/voice-lounge/src/esphome/components/captive_portal/captive_portal.o
Compiling .pioenvs/voice-lounge/src/esphome/components/display/display.o
Compiling .pioenvs/voice-lounge/src/esphome/components/display/display_buffer.o
Compiling .pioenvs/voice-lounge/src/esphome/components/display/rect.o
Compiling .pioenvs/voice-lounge/src/esphome/components/esp32/core.o
Compiling .pioenvs/voice-lounge/src/esphome/components/esp32/gpio.o
Compiling .pioenvs/voice-lounge/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/voice-lounge/src/esphome/components/esp_adf/esp_adf.o
Compiling .pioenvs/voice-lounge/src/esphome/components/esp_adf/microphone/esp_adf_microphone.o
Compiling .pioenvs/voice-lounge/src/esphome/components/esp_adf/speaker/esp_adf_speaker.o
Compiling .pioenvs/voice-lounge/src/esphome/components/esphome/ota/ota_esphome.o
Compiling .pioenvs/voice-lounge/src/esphome/components/factory_reset/button/factory_reset_button.o
src/esphome/components/esp_adf/microphone/esp_adf_microphone.cpp: In static member function 'static void esphome::esp_adf::ESPADFMicrophone::read_task(void*)':
src/esphome/components/esp_adf/microphone/esp_adf_microphone.cpp:110:3: warning: missing initializer for member 'i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
   };
   ^
src/esphome/components/esp_adf/microphone/esp_adf_microphone.cpp:110:3: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
src/esphome/components/esp_adf/microphone/esp_adf_microphone.cpp:110:3: warning: missing initializer for member 'i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
src/esphome/components/esp_adf/microphone/esp_adf_microphone.cpp:110:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
src/esphome/components/esp_adf/microphone/esp_adf_microphone.cpp:110:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
src/esphome/components/esp_adf/microphone/esp_adf_microphone.cpp:110:3: warning: missing initializer for member 'i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
src/esphome/components/esp_adf/speaker/esp_adf_speaker.cpp: In static member function 'static void esphome::esp_adf::ESPADFSpeaker::player_task(void*)':
[... cut ...]
Compiling .pioenvs/voice-lounge/components/esp-dsp/modules/common/misc/dsps_pwroftwo.o
Compiling .pioenvs/voice-lounge/components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.o
Compiling .pioenvs/voice-lounge/components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_m_ae32.o
xtensa-esp32s3-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/voice-lounge/src/main.o] Error 1
========================= [FAILED] Took 128.90 seconds =========================

BigBobbas, 2024-12-21:
@Clasper101 try doing a clean build files - from the 3 dot menu of the device card in the ESPHome dashboard and then try to recompile

Mis1ify, 2024-12-21:
Nope, cleaning build files does not work. I'm also experiencing the problem on newest 2024.12.2 release, cleaned build files and restarted the add-on, host os, etc.

Clasper101, 2024-12-21:
With the extra yaml code from above and a reboot I have managed to update my Atom Echoes and S3 Box 3.

fortuna, 2024-12-24:
The proposed YAML in https://github.com/esphome/wake-word-voice-assistants/issues/42#issuecomment-2547515639 did not fix it for me.

I'm getting the same error as @Mis1ify in https://github.com/esphome/wake-word-voice-assistants/issues/42#issuecomment-2558129867, and also have the latest esphome 2024.12.2.

No workaround yet. I wanted to revert to an older version, but unfortunately there doesn't seem to be a way to install a previous version of esphome with brew:

```console
$ brew install esphome@2024.12.0
Warning: No available formula with the name "esphome@2024.12.0".
==> Searching for similarly named formulae and casks...
Error: No formulae or casks found for esphome@2024.12.0.
```

```console
$ brew search esphome@
==> Formulae
esphome ✔

==> Casks
epichrome
```

fortuna, 2024-12-24:
Reverting back to 2024.11.3 fixed the issue for me.

I figured out how to revert to a previous version on Homebrew. I had to create a tap and extract the old version:

```sh
brew uninstall esphome
brew tap-new fortuna/esphome-versions
brew extract --version 2024.11.3 esphome fortuna/esphome-versions
brew install fortuna/esphome-versions/esphome@2024.11.3
```

Credit: https://stackoverflow.com/a/64125796/1234111

@jesserockz would it make sense to revert 2024.12?

michaelkleinhenz, 2025-02-24:
> To use the s3 boxes with ESPHome 2024.12 or above will require some reworking of the audio drivers due to the craziness of esp-adf and esp-idf. Work has begun on this to remove the esp-adf dependency but is not complete yet. There is no timeline available.
> 
> Jesse

@jesserockz is this still ongoing? Is there a link? Thanks..

BigBobbas, 2025-02-24:
@michaelkleinhenz there is update firmware now available https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box-3/esp32-s3-box-3.yaml

jesserockz, 2025-02-24:
As above, the firmwares in this repo have been updated for ESPHome 2024.2.0 without using esp-adf now

closed 2025-02-24 by jesserockz

jimmyeao, 2025-03-05:
m5stack-atom-echo still uses IDF...

> As above, the firmwares in this repo have been updated for ESPHome 2024.2.0 without using esp-adf now