esphome/wake-word-voice-assistants #43: 2024.12.0 Error Upgrade Firmware with HASSIO Esphome

voyagerft, 2024-12-18:
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-9f9bec.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5stack-atom-echo-9f9bec (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/m5stack-atom-echo-9f9bec/.pioenvs/m5stack-atom-echo-9f9bec/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 3.99 seconds ==========================

tradosh, 2024-12-18:
adding this in esp32 has solved the issue (loads of warrning in the build still though)
```
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

rkitzman, 2024-12-21:
> adding this in esp32 has solved the issue (loads of warrning in the build still though)
> 
> ```
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```

Confirmed with my install.

FeroX-Ron, 2024-12-21:
where to add?

rkitzman, 2024-12-21:
> where to add?

Add to the esp32: section

ballakers, 2024-12-23:
Still an issue in 12.2 compiler and happens with the esp32-s3-box-3, didn't try on the atom echo but assume the same.  Weird how can flash fine from the web-flasher and then can't install an OTA update via HA?  These things are finicky for sure, maybe one day it'll all get figured out nicely.

dro-ex, 2024-12-28:
Still an issue in 12.2 compiler with atom echo

profucius, 2025-01-16:
Chiming in to say I also had the issue from OP today. ESPhome 2024.12.2 with ESP32-Box-S3-Lite.
Adding the lines by tradosh above allowed it to proceed, with many warnings during build.

ballakers, 2025-01-17:
Guessing they are coming out with a major Feb release based on the slow December releases still coming out and the release of VoicePE will probably make major changes for this voice assistant stuff, hopefully making it easier to modify these readymade configs to make our own - we all know it's been a pain to make use of the extra sensors that comes with the s3-box-3 with dock etc. Here's to hoping!