esphome/wake-word-voice-assistants #45: Atom Echo M5 install failure

tradosh, 2024-12-18:
if I try through website (https://esphome.io/projects/index.html) it works fine, however when trying to upload config to the ESPHome for hte install I'm getting following error:

```INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/esphome-web-0a06ec.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing esphome-web-0a06ec (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-0a06ec/.pioenvs/esphome-web-0a06ec/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.74 seconds ==========================```

alvinchen1, 2024-12-18:
Looks like it might be the same error as https://github.com/esphome/wake-word-voice-assistants/issues/43

tradosh, 2024-12-19:
just a note that even after adding the workaround, when trying to use the local wake word getting:
[10:23:07][E][micro_wake_word:032]: Could not allocate the streaming model's tensor arena.
[10:23:07][E][micro_wake_word:268]: Failed to initialize a wake word model.
[10:23:07][E][micro_wake_word:161]: Failed to load the wake word model(s) or allocate buffers
[10:23:07][E][component:164]: Component micro_wake_word set Error flag: unspecifie****

tradosh, 2024-12-19:
worked out that using the bluetooth_proxy causes that to happen, if I disable bluetooth_proxy it starts workign fine even with local wake word - would be good to see if we can edit the code to allow the bluetooth proxy to be used with it

LeVraiRoiDHyrule, 2025-03-14:
Hi, 

I am begginer in esphome. I too am trying to add the bluetooth proxy feature to my atom echo satellite. My configuration is the following:
```
substitutions:
  name: m5stack-atom-echo-0f1130
  friendly_name: AtomEchoLit
  micro_wake_word_model: okay_nabu  # alexa, hey_jarvis, hey_mycroft are also supported

esphome:
  min_version: 2025.2.0
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0

logger:

ota:
  - platform: esphome
    id: ota_esphome

wifi:
  ap:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    bits_per_sample: 32bit
    channel: right
    buffer_duration: 60ms

media_player:
  - platform: speaker
    name: None
    id: echo_media_player
    announcement_pipeline:
      speaker: echo_speaker
      format: WAV
    codec_support_enabled: false
    buffer_size: 6000
[... 267 lines cut ...]
          then:
            - lambda: id(va).set_use_wake_word(false);
            - voice_assistant.stop
            - delay: 500ms
            - micro_wake_word.start

micro_wake_word:
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
  vad:
  models:
    - model: ${micro_wake_word_model}

bluetooth_proxy:
  active: true
```

Without `bluetooth_proxy:
  active: true`, it compiles without issue. But with it, the total size of the program is greater than the maximum available storage on the device.

Is there features in my configuration that I can remove to save some space without compromising too much on features ?
Did you succeed to have a echo voice assistant that is also a bluetooth_proxy on atom echo ?

Thanks in advance for any answer, have a nice day

alvinchen1, 2025-03-14:
Adding Bluetooth Proxy to Voice Assistant might not work.  See https://esphome.io/components/voice_assistant

> Warning
> Audio and voice components consume a significant amount of resources (RAM, CPU) on the device.
> 
> Crashes are likely to occur if you include too many additional components in your device’s configuration. In particular, Bluetooth/BLE components are known to cause issues when used in combination with Voice Assistant and/or other audio components.