esphome/wake-word-voice-assistants #48: M5Stack Atom Echo does not play timer_finished

leilich, 2024-12-22:
Installed a fresh Atom Echo via https://esphome.io/projects/ as Voice Assistant. (Tried also with 2 other Echoes.)
Firmware: 24.11.1 (ESPHome 2024.11.1)
Integrated in HomeAssistant, not yet adopted by ESPhome Add-On.
Similiar behaviour if adopted to ESPhome and changed the versions resulting in Firmware: 2024.12.2.
```
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```
Wake-Word-Detection and Assist works fine.
Setting a timer for 1sec or 2sec ("Timer für eine Sekunde") lets play the echo the timer_finished sound and flash green. The sound is played while assist answers.
Setting a timer for more than 2sec plays no sound but flashes green.

jaycollett, 2024-12-24:
Same issue. I had to apply a workaround to get the latest version of Home Assistant even to compile the firmware:

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0

One-second timers work, but anything longer fails to play any sound; I see the logs showing the speaker stopping and starting, but I do not get audio output. I do get the green-blinking LEDs.

leilich, 2024-12-28:
Tested a bit more: seems that the voice-assistant component locks the media player.
The media_player from the fresh atom echo via https://esphome.io/projects/ including voice assistant didn't play any media started from HA either.
In this test I disabled the voice assistant (commented it out in the yaml) so there remains only the media player and the button; now the media player plays media as expected. Log comes as expected:
```
[21:34:03][D][media_player:065]: 'M5Stack Atom Echo Küche' - Setting
[21:34:03][D][media_player:072]:   Media URL: http://192.168.178.78:8123/media/local/bla.mp3...
```

AlexanderLeporiday, 2025-01-12:
I don't have a suggestion on how to fix this yet but I am getting the same issue as well.
I've had to update the framework section to include a version and platform_version, despite this not being recommended it's the only way to get CMake to compile it.

The audio seems to be unable to play when the timer triggers because the media player is disabled after the voice assistant finishing speaking.

However the media player and voice assistant can talk at the same time because the timer goes off while it's still speaking.

PR #60 (2025-02-20): [M5 Echo] Use new media player to support announcements
Uses the new speaker media player in ESPHome 2025.2 for handling the voice assistants audio output. It still is built with esp-idf 4.4.8 because the PDM microphone doesn't seem to work on 5.1.5.

The media player is in its most basic form, as it only supports wav files. Buffers are kept small to conserve memory. The timer takes advantage of the new media player's repeat one mode for the timer beep, but it currently requires a lambda to set.

Closes #48 (uses the new media player code to play the timer finished sound)
Closes #54 (updated v [... cut]
closed 2025-02-20 by kbx81