esphome/wake-word-voice-assistants #50: stop timer via voice after it finished

pohly, 2025-01-04:
I am just starting with Home Assistant voice assistant using the ESP32-S3-BOX. My initial use case is for setting timers in the kitchen. I got voice commands for that working, but then found that when a timer fires, I have to use the button on the device to acknowledge the timer. Voice commands don't work while a timer alarm is playing.

This matches the configuration:
https://github.com/esphome/wake-word-voice-assistants/blob/35f6e76cfa86bc7dc22058ccda1cdb0eb81383ff/esp32-s3-box/esp32-s3-box.yaml#L237-L256

This is from https://github.com/esphome/wake-word-voice-assistants/commit/d125927e7ba7e14d9eb615d068474ebda5566dea => cc @jlpouffier

Can this be changed so that the voice assistant remains active while playing the alarm? I intend to place the device in the kitchen where it is out of the way and thus hard to reach. Besides, the main purpose of voice assistant is to have the hands free for other tasks... :sweat_smile:

jlpouffier, 2025-01-13:
It is indeed the reason.
The fact of using the microphones while the speaker is active requires a fair bit of audio processing, echo cancelation, and all that to make it a useful experience.

So we are allowing it on the Voice PReview Edition, because we do process the audio and cancel the echo, but not on the box, unfortunately.

closed 2025-01-13 by jlpouffier

pohly, 2025-01-13:
A more extensive state tracking could solve this: when a timer is finished, set a flag that [periodically](https://esphome.io/components/interval.html) causes the beep to play, but never at the same time as voice output or while capturing audio. There also has to be a long enough pause after a beep to allow saying the wake word.

I might give this a try...