esphome/wake-word-voice-assistants #58: Atom Echo voice assistant cannot be installed manually

Epicek1, 2025-01-24:
I wanted to modify the code for the Atom Echo voice assistant, so I tried to install the full code via the ESPhome builder (instead of the default one, that is installed from the "$13 voice assistant for Home Assistant" guide, and just refers to the github repository). The installation failed many times, it gives this error message. What am I missing? 

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/m5stack-atom-echo-a16950.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5stack-atom-echo (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Configuring incomplete, errors occurred!
See also "/data/build/m5stack-atom-echo/.pioenvs/m5stack-atom-echo/CMakeFiles/CMakeOutput.log".
See also "/data/build/m5stack-atom-echo/.pioenvs/m5stack-atom-echo/CMakeFiles/CMakeError.log".

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

========================= [FAILED] Took 10.43 seconds =========================
```

Badn1nja, 2025-02-11:
`substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu  # alexa, hey_jarvis, hey_mycroft are also supported

esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
  min_version: 2024.9.0

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "123213213123123123"

ota:
  - platform: esphome
    password: "12312312312311232312"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Stack-Atom-Echo"
    password: "12312131223123"
`   I changed this in my yaml (yaml was copied directly from repo) , Went to 'ESPHome Builder', > install > Manual Download. then flashed m5stack-atom-echo.factory.bin via ESPHome.Web. Adopted the ESP with API key. This got it working!

jesserockz, 2025-02-27:
Can you retry with the latest YAML configuration located in this repo? It has been significantly changed since the last ESPHome release to remove the esp-adf dependency.

FoliniC, 2025-02-28:
quick version:
new yaml didn't work with my esp32s3, tried to merge with my previous modirication. It's deployed correctly but it didn't work. Next week I'll try to understand why.
The  "working" version:
`substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu  # alexa, hey_jarvis, hey_mycroft are also supported

esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
  min_version: 2025.2.0

esp32:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
  board: esp32-s3-devkitc-1
logger:
  level: VERY_VERBOSE
api:
  encryption:
    key: "DSSSSSSSSSSSSSSSSSSSSSSSSSSSSSw="

ota:
  - platform: esphome
    id: ota_esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

captive_portal:

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO13
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO4
    adc_type: external  
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO5
    dac_type: external
    bits_per_sample: 32bit
    channel: right
    buffer_duration: 60ms

media_player:
  - platform: speaker
    name: None
    id: echo_media_player
    announcement_pipeline:
      speaker: echo_speaker
      format: WAV
[... 293 lines cut ...]
Configuration does not match the platform of the connected device. Expected an ESP32 device.

I'm using the correct repo?
My device is labelled: m5stack ATOM S3 LITE ESP32-S3FN8
when compiling I got: ESP32S3 240MHz, 320KB RAM, 8MB Flash

The version I tried to modify differs from the repo in 
my> board: esp32-s3-devkitc-1
repo> board: m5stack-atom

my> i2s_lrclk_pin: GPIO13
repo> i2s_lrclk_pin: GPIO33

my> i2s_din_pin: GPIO4
repo> i2s_din_pin: GPIO23

my> i2s_dout_pin: GPIO5
repo> i2s_dout_pin: GPIO22

repo> introduced media_player

my> pin: GPIO14
repo> pin: GPIO27

my> use external_components:

FoliniC, 2025-03-05:
Double checked the problem, it seems that I was using [ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/index.html) which is for WROOM.
The right one should be [ESP32-S3-DevKitM-1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitm-1/index.html), but ESPHome didn't find it.
`INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/cpe-24g-112e90_downloaded.yaml...
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING This board is unknown. Make sure the chosen chip component is correct.
substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu
esphome:
  name: m5stack-atom-echo
  name_add_mac_suffix: true
  friendly_name: M5Stack Atom Echo
  min_version: 2025.2.0
  build_path: build/m5stack-atom-echo
  area: ''
  platformio_options: {}
  includes: []
  libraries: []
esp32:
  framework:
    version: 5.1.5
    sdkconfig_options: {}
    advanced:
      ignore_efuse_custom_mac: false
    components: []
    platform_version: https://github.com/pioarduino/platform-espressif32.git#51.03.07
    source: pioarduino/framework-espidf@https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
    type: esp-idf
  variant: ESP32S3
  board: ESP32-S3-DevKitM-1
  flash_size: 4MB`

Is it the right board to set for my ESP device? How can I check if they match?

FoliniC, 2025-03-05:
A little step in the right direction! Now I can flash my device and it's discovered by home assistant!
When I click on "add" I got:
Discovered ESPHome node
Do you want to add the ESPHome node M5Stack Atom Echo 5cedac to Home Assistant?

Invalid flow specified

Any idea on how to solve this?

My "macumba" YAML:

`substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu  # alexa, hey_jarvis, hey_mycroft are also supported

esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
  min_version: 2025.2.0

esp32:
  framework:
    type: esp-idf
    # version: 4.4.8
    # platform_version: 5.4.0
  board: esp32-s3-devkitc-1
logger:
  level: VERY_VERBOSE
api:
  encryption:
    key: "ssssssssssssssssssssssssssssssssssssssssss="

ota:
  - platform: esphome
    id: ota_esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    # Optional manual IP configuration
  manual_ip:
    static_ip: 192.168.0.222
    gateway: 192.168.0.100
    subnet: 255.255.255.0
    dns1: 8.8.8.8
    dns2: 8.8.4.4

captive_portal:

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO13
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO4
    adc_type: external  
    pdm: true

speaker:
[... 271 lines cut ...]
    on_value:
      - if:
          condition:
            lambda: return x == "In Home Assistant";
          then:
            - micro_wake_word.stop
            - delay: 500ms
            - lambda: id(va).set_use_wake_word(true);
            - voice_assistant.start_continuous:
      - if:
          condition:
            lambda: return x == "On device";
          then:
            - lambda: id(va).set_use_wake_word(false);
            - voice_assistant.stop
            - delay: 500ms
            - micro_wake_word.start

micro_wake_word:
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
  vad:
  models:
    - model: ${micro_wake_word_model}`

Badn1nja, 2025-03-10:
I was still have issues but I managed to get it working by using the easy install on the article..

Thank you @jesserockz. Really awesome work I've managed to get it working with Mistral 70b, Whisper and Piper and it think I might be able to start getting rid of the useless Google home!

closed 2025-04-30 by jesserockz