esphome/wake-word-voice-assistants #59: M5 Stack Atom Echo_tinny sound

JanKraslice, 2025-02-12:
Hello, I have several M5 after some update, instead of clear sound ( I use Czech) echo with M5 tinny echo.
It is hard to describe so I am attaching the voice output recorded by the phone it is an announcement of what the weather will be today:

m5 voice
Home Assistant Voice Preview Edition
As you can hear the tinny voice problem is only on the M5!
I updated my m5 to the latest version of ESP and HM also and nothing still the same tinny sound on my M5

M5:Firmware: 2024.12.4
CORE:2025.1.4
SYS:14.2
I can't embed the audio here so please find the download link here:
[[I can't embed the audio here so please find the download link here](https://www.uschovna.cz/zasilka/RIKAHGP7P5ADKKNY-NRG/)](url)
Thank you for your help

BelenCaSa, 2025-02-15:
Hello
The same thing happens to me since esphome version 2024.11.0
I have left my Atom with version 2024.10, any higher version makes the metallic voice sound.
I have also observed that when the metallic voice is played, the end of timer warning is not played either, the LED does flash green.

JanKraslice, 2025-02-18:
Please do I BelenCaSa's the only two having this problem? Isn't this addressed somewhere in another discussion?Alternatively, how do I get back to version 2024.10Thank you
-------- Původní zpráva --------Od: Belen ***@***.***> Datum: 15.02.25  6:29  (GMT+01:00) Komu: esphome/wake-word-voice-assistants ***@***.***> Cc: JanKraslice ***@***.***>, Author ***@***.***> Předmět: Re: [esphome/wake-word-voice-assistants] M5 Stack Atom Echo_tinny sound (Issue #59) 
  Hello
The same thing happens to me since esphome version 2024.11.0
I have left my Atom with version 2024.10, any higher version makes the metallic voice sound.
I have also observed that when the metallic voice is played, the end of timer warning is not played either, the LED does flash green.

Hello
The same thing happens to me since esphome version 2024.11.0
I have left my Atom with version 2024.10, any higher version makes the metallic voice sound.
I have also observed that when the metallic voice is played, the end of timer warning is not played either, the LED does flash green.

BelenCaSa, 2025-02-18:
Hello Jan Kraslice

I keep all the firmware of all my devices on my laptop with the version and date, because if when updating a device, the new version has an error, I can install the previous version. I also find it strange that only you and I have this problem, let's wait.

PR #60 (2025-02-20): [M5 Echo] Use new media player to support announcements
Uses the new speaker media player in ESPHome 2025.2 for handling the voice assistants audio output. It still is built with esp-idf 4.4.8 because the PDM microphone doesn't seem to work on 5.1.5.

The media player is in its most basic form, as it only supports wav files. Buffers are kept small to conserve memory. The timer takes advantage of the new media player's repeat one mode for the timer beep, but it currently requires a lambda to set.

Closes #48 (uses the new media player code to play the timer finished sound)
Closes #54 (updated v [... cut]
closed 2025-02-20 by kbx81