esphome/wake-word-voice-assistants #66: Atom Echo will not compile with ESPHome 2025.2.0

arjen-w, 2025-02-19:
I upgraded ESPHome to version 2025.2.0 and I try to upgrade my Atom Echo M5 speaker.
However, I get the following errors while compiling:

`INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-104c50.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing speaker-werkkamer (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
Library Manager: Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
INFO Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
Unpacking  [####################################]  100%
Library Manager: ESPMicroSpeechFeatures@1.1.0 has been installed!
INFO ESPMicroSpeechFeatures@1.1.0 has been installed!
Library Manager: Installing esphome/esp-audio-libs @ 1.1.1
INFO Installing esphome/esp-audio-libs @ 1.1.1
Unpacking  [####################################]  100%
Library Manager: esp-audio-libs@1.1.1 has been installed!
INFO esp-audio-libs@1.1.1 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ESPMicroSpeechFeatures @ 1.1.0
|-- esp-audio-libs @ 1.1.1
Generating assembly for .pioenvs/speaker-werkkamer/duer_profile.S
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/api_connection.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/api_server.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/list_entities.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/proto.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/api/user_services.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/audio/audio.o
Compiling .pioenvs/speaker-werkkamer/src/esphome/components/audio/audio_decoder.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
[... 35 lines cut ...]
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
src/esphome/components/audio/audio_decoder.cpp:321:29: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                             ^~
src/esphome/components/audio/audio_decoder.cpp:321:68: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                    ^~
src/esphome/components/audio/audio_decoder.cpp:321:104: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                                                        ^~
src/esphome/components/audio/audio_decoder.cpp:323:49: error: base operand of '->' is not a pointer
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
*** [.pioenvs/speaker-werkkamer/src/esphome/components/audio/audio_decoder.o] Error 1
========================= [FAILED] Took 17.50 seconds =========================
`

I have copied the config code from: [https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml] and a validation of the code is succesful.

voyagerft, 2025-02-19:
same problem

cooljimy84, 2025-02-19:
using the same config i can build on 2024.12, fails with the same error above on 2025.2.0

jonasclaes, 2025-02-19:
Experiencing the same as other users already mentioned in this thread.

```
INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-214584.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5stack-atom-echo-214584 (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ESPMicroSpeechFeatures @ 1.1.0
|-- esp-audio-libs @ 1.1.1
Compiling .pioenvs/m5stack-atom-echo-214584/src/esphome/components/audio/audio_decoder.o
Compiling .pioenvs/m5stack-atom-echo-214584/src/esphome/components/audio/audio_resampler.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                                                          ^
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 2 is invalid
src/esphome/components/audio/audio_decoder.cpp: In member function 'esp_err_t esphome::audio::AudioDecoder::start(esphome::audio::AudioFileType)':
src/esphome/components/audio/audio_decoder.cpp:79:40: error: 'esp_audio_libs' was not declared in this scope
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                        ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:28: error: parse error in template argument list
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:81: error: no matching function for call to 'make_unique<<expression error> >()'
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
In file included from src/esphome/components/speaker/speaker.h:12,
                 from src/esphome/components/audio/audio_transfer_buffer.h:8,
                 from src/esphome/components/audio/audio_decoder.h:6,
                 from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/core/helpers.h:85:59: note: candidate: 'template<class T, class ... Args> std::unique_ptr<T> esphome::make_unique(Args&& ...)'
 template<typename T, typename... Args> std::unique_ptr<T> make_unique(Args &&...args) {
                                                           ^~~~~~~~~~~
src/esphome/core/helpers.h:85:59: note:   template argument deduction/substitution failed:
src/esphome/components/audio/audio_decoder.cpp:79:81: error: template argument 1 is invalid
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
src/esphome/components/audio/audio_decoder.cpp:80:25: error: base operand of '->' is not a pointer
       this->wav_decoder_->reset();
                         ^~
src/esphome/components/audio/audio_decoder.cpp: In member function 'esphome::audio::FileDecoderState esphome::audio::AudioDecoder::decode_wav_()':
src/esphome/components/audio/audio_decoder.cpp:314:5: error: 'esp_audio_libs' has not been declared
     esp_audio_libs::wav_decoder::WAVDecoderResult result = this->wav_decoder_->decode_header(
     ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:9: error: 'result' was not declared in this scope
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
[... cut ...]
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
*** [.pioenvs/m5stack-atom-echo-214584/src/esphome/components/audio/audio_decoder.o] Error 1
========================== [FAILED] Took 3.03 seconds ==========================
```

wesselah, 2025-02-19:
Same problem even timestamp is identical

awaldram, 2025-02-20:
`substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu  # alexa, hey_jarvis, hey_mycroft are also supported

esphome:
  name: ${name}
  name_add_mac_suffix: True
  friendly_name: ${friendly_name}
  min_version: 2025.2.0

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0

logger:
api:
 

ota:
  - platform: esphome
    id: ota_esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    bits_per_sample: 16bit
    channel: right
    buffer_duration: 60ms

#media_player:
#  - platform: speaker
#    name: None
#    id: echo_media_player
#    announcement_pipeline:
#      speaker: echo_speaker
#      format: WAV
#    codec_support_enabled: false
#    buffer_size: 5000
#    files:
#      - id: timer_finished_wave_file
#        file: https://github.com/esphome/wake-word-voice-assistants/raw/main/sounds/timer_finished.wav
[... 257 lines cut ...]
          then:
            - micro_wake_word.stop
            - delay: 500ms
            - lambda: id(va).set_use_wake_word(true);
            - voice_assistant.start_continuous:
      - if:
          condition:
            lambda: return x == "On device";
          then:
            - lambda: id(va).set_use_wake_word(false);
            - voice_assistant.stop
            - delay: 500ms
            - micro_wake_word.start

micro_wake_word:
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
  vad:
  models:
    - model: ${micro_wake_word_model}`

This should compile and work ..(it does for me) removed all the mediaplayer stuff (as this was hard crashing the Atom. re-added speaker as VA output .

Probably breaks timer but it should work as a stop gap. you may need to clean build to clear the compile error.

awaldram, 2025-02-20:
```
substitutions:
  name: m5stack-atom-echo-9f6754
  friendly_name: M5Stack Atom Echo-9f6754
  micro_wake_word_model: okay_nabu  # alexa, hey_jarvis, hey_mycroft are also supported

esphome:
  name: ${name}
  name_add_mac_suffix: False
  friendly_name: ${friendly_name}
  min_version: 2025.2.0

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0

logger:
api:
  encryption:
    key: IS4YJXUvlZkyGkpOUNjlkYvjCmYDPdoJaex19eHwLJU=

ota:
  - platform: esphome
    id: ota_esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    bits_per_sample: 16bit
    channel: right
    buffer_duration: 60ms

#media_player:
#  - platform: speaker
#    name: None
#    id: echo_media_player
#    announcement_pipeline:
#      speaker: echo_speaker
#      format: WAV
#    codec_support_enabled: false
#    buffer_size: 5000
#    files:
[... 264 lines cut ...]
      - if:
          condition:
            lambda: return x == "On device";
          then:
            - lambda: id(va).set_use_wake_word(false);
            - voice_assistant.stop
            - delay: 500ms
            - micro_wake_word.start

micro_wake_word:
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
  vad:
  models:
    - model: ${micro_wake_word_model}
```

Removed Mediaplayer stuff as it doesn't work anyway (at least for me crashes the Atom stack overflow). probably breaks timer and doesn't fix Button control.

But it should at least function as a voice asisst node and may help some with broken M5's

You made need to 'build clean' to clear the audio compile error. 

No warranty just thought it may help some having issues.

chpego, 2025-02-20:
Despite [the new version released](https://github.com/esphome/wake-word-voice-assistants/releases/tag/25.2.1) a few hours ago, the compilation still does not compile, even after doing a "clean build files" with [ESPHome](https://github.com/esphome/esphome/releases/tag/2025.2.0)

jonasclaes, 2025-02-20:
@chpego, can you try changing `@main` to `@389be0bf1dd5e45594ac645aa4cf823c771c04b3`? That's the specific commit id, and that did the trick for me. Caching might be throwing a wrench into the wheel here.

chpego, 2025-02-20:
> [@chpego](https://github.com/chpego), can you try changing `@main` to `@389be0bf1dd5e45594ac645aa4cf823c771c04b3`? That's the specific commit sha, and that did the trick for me. Caching might be throwing a wrench into the wheel here.

ok it works but I don't understand where the code is? sorry I'm a bit ignorant on this subject.

mesoproterozoic, 2025-02-20:
compiles now for me, even with `@main`

closed 2025-02-20 by arjen-w