esphome/wake-word-voice-assistants #70: Wake word greyed out?

starman2k01, 2025-02-20:
I'm on 25.2.1; same issue on previous release. Installed firmware via 
https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/
Under ESPHome integration I see:

[image]
I can change "Wake word engine location" from "On device" to "In Home Assistant". 
But the Wake word item is greyed out. I'd like to change the default microwakeword from "Hello Nabu" to "hey jarvis". I think I need a custom firmware to do that, but why is the "Wake word" pulldown menu grey?

simowilso, 2025-02-22:
Yes I'd like to know the recommended method for selecting a different wakeword too please

starman2k01, 2025-02-23:
I went ahead and took over the device with ESPHome device builder, copied the yaml code from here:
https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml
changed the wake word in the yaml and boom there you go.

starman2k01, 2025-02-23:
I still have no idea what the greyed-out wake word pulldown menu item is supposed to be!

simowilso, 2025-02-23:
Yes I ended up doing the esphome adoption as well. It's tough to keep up with all the changes going on when you don't play with it every day ;)

closed 2025-05-08 by github-actions[bot]