esphome/wake-word-voice-assistants #71: M5 Atom Echo - crash on Announce

brettmiller, 2025-02-21:
After https://github.com/esphome/wake-word-voice-assistants/pull/60 the Atom Echoes crash when Announcing.
```
[D][media_player:086]:  Announcement: yes
[D][voice_assistant:506]: State changed from IDLE to STREAMING_RESPONSE
[D][voice_assistant:512]: Desired state set to STREAMING_RESPONSE
[D][light:036]: 'M5Stack Atom Echo 80472c' Setting:
[D][light:051]:   Brightness: 100%
[D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[D][speaker_media_player:420]: State changed to ANNOUNCING

[23:01:14]***ERROR*** A stack overflow in task ann_read has been detected.
[23:01:14]
[23:01:14]
[23:01:14]Backtrace: 0x400833ed:0x3ffe4a70 0x4008ce89:0x3ffe4a90 0x4009022d:0x3ffe4ab0 0x4008ebb5:0x3ffe4b30 0x4008cf84:0x3ffe4b60 0x4008cf36:0x3ffb4210 0x733ffffc:0x00000000 |<-CORRUPTED
[23:01:14]
[23:01:14]
[23:01:14]
[23:01:14]
[23:01:15]ELF file SHA256: 05d7e199ca7540d4
[23:01:15]
[23:01:15]Rebooting...
```

I've test two Echoes, one setup with `esphome` locally with a yaml that includes w/ a remote_package and another setup via the ready made project on the esphome.io website. Both have the issue. Per comments on the PR it seems others are have this issue as well.

awaldram, 2025-02-21:
Full cycle from boot to 
1. OK Nabu
2. Whats the time
3. Time found and ready to speak 'wav'
4. start Media player 'announce'
5. full hard crash
6. rebooting
7. ready to crash again ::

```

[D][esp-idf:000][update_task]: I (18690) esp-x509-crt-bundle: Certificate validated

[D][update:011][update_task]: 'Firmware' - Publishing:
[D][update:012][update_task]:   Current Version: 25.2.1
[D][update:015][update_task]:   Latest Version: 25.2.1
[D][update:018][update_task]:   Firmware URL: https://firmware.esphome.io/wake-word-voice-assistant/m5stack-atom-echo/25.2.1/m5stack-atom-echo-esp32.ota.bin
[D][update:021][update_task]:   Title: m5stack.atom-echo-wake-word-voice-assistant
[D][update:023][update_task]:   Summary: ## What's Changed
[09:28:51]* Bump esphome/workflows from 2024.11.1 to 2024.11.3 by @dependabot in https://github.com/esphome/wake-word-voice-assistants/pull/37
[09:28:51]* Bump esphome/workflows from 2024.11.3 to 2024.12.0 by @dependabot in https://github.com/esphome/wake-word-voice-assistants/pull/40
[09:28:51]* Bump esphome/workflows from 2024.12.0 to 2025.1.0 by @dependabot in https://github.com/esphome/wake-word-voice-assistants/pull/55
[09:31:51]* define esp-idf version to 4
[D][update:026][update_task]:   Release URL: https://github.com/esphome/wake-word-voice-assistants/releases/tag/25.2.1
[D][api:103]: Accepted 192.168.0.10
[W][component:170]: Component api cleared Warning flag
[D][api.connection:1446]: Home Assistant 2025.2.4 (192.168.0.10): Connected successfully
[W][micro_wake_word:183]: Wake word is already stopped
[D][light:036]: 'M5Stack Atom Echo 9f6754' Setting:
[D][light:051]:   Brightness: 60%
[D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[D][ring_buffer:034]: Created ring buffer with size 2048
[D][micro_wake_word:399]: Resetting buffers and probabilities
[D][micro_wake_word:195]: State changed from IDLE to START_MICROPHONE
[W][component:237]: Component script took a long time for an operation (61 ms).
[W][component:238]: Components should block for at most 30 ms.
[D][micro_wake_word:107]: Starting Microphone
[D][micro_wake_word:195]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[D][esp-idf:000]: I (31705) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[D][micro_wake_word:195]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[D][micro_wake_word:162]: The 'Okay Nabu' model sliding average probability is 0.980 and most recent probability is 1.000
[D][micro_wake_word:123]: Wake Word 'Okay Nabu' Detected
[D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[D][micro_wake_word:129]: Stopping Microphone
[D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[D][esp-idf:000]: I (197191) I2S: DMA queue destroyed

[D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE
[D][voice_assistant:506]: State changed from IDLE to START_MICROPHONE
[D][voice_assistant:512]: Desired state set to START_PIPELINE
[D][voice_assistant:223]: Starting Microphone
[D][ring_buffer:034]: Created ring buffer with size 16384
[D][voice_assistant:506]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[D][esp-idf:000]: I (197243) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[D][voice_assistant:506]: State changed from STARTING_MICROPHONE to START_PIPELINE
[D][voice_assistant:277]: Requesting start...
[D][voice_assistant:506]: State changed from START_PIPELINE to STARTING_PIPELINE
[D][voice_assistant:527]: Client started, streaming microphone
[D][voice_assistant:506]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[D][voice_assistant:512]: Desired state set to STREAMING_MICROPHONE
[D][voice_assistant:642]: Event Type: 1
[D][voice_assistant:645]: Assist Pipeline running
[D][voice_assistant:642]: Event Type: 3
[D][voice_assistant:656]: STT started
[D][light:036]: 'M5Stack Atom Echo 9f6754' Setting:
[... 122 lines cut ...]
[D][select:015]: 'Wake word engine location': Sending state On device (index 1)
[D][esp-idf:000]: I (275) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[D][binary_sensor:034]: 'Button': Sending initial state OFF
[C][light:021]: Setting up light 'M5Stack Atom Echo 9f6754'...
[D][light:036]: 'M5Stack Atom Echo 9f6754' Setting:
[D][light:041]:   Color mode: RGB
[D][template.switch:046]:   Restored state ON
[D][switch:012]: 'Use listen light' Turning ON.
[D][switch:055]: 'Use listen light': Sending state ON
[D][light:036]: 'M5Stack Atom Echo 9f6754' Setting:
[D][light:047]:   State: ON
[D][light:051]:   Brightness: 60%
[D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[D][template.switch:046]:   Restored state OFF
[D][switch:016]: 'timer_ringing' Turning OFF.
[D][switch:055]: 'timer_ringing': Sending state OFF
[D][media_player:073]: 'M5Stack Atom Echo 9f6754' - Setting
[D][media_player:077]:   Command: REPEAT_OFF
[D][media_player:086]:  Announcement: yes
[D][media_player:073]: 'M5Stack Atom Echo 9f6754' - Setting
[D][media_player:077]:   Command: STOP
[D][media_player:086]:  Announcement: yes
[C][i2s_audio:028]: Setting up I2S Audio...
```

slackerbob, 2025-02-21:
I can confirm the same thing:

```
[07:11:37][D][media_player:073]: 'Echo Assistant - Bathroom a02af0' - Setting
[07:11:37][D][media_player:080]:   Media URL: https://redacted/api/tts_proxy/5k8-zfpo3ncbRhHG1uqCpg.wav
[07:11:37][D][media_player:086]:  Announcement: yes
[07:11:37][D][voice_assistant:506]: State changed from IDLE to STREAMING_RESPONSE
[07:11:37][D][voice_assistant:512]: Desired state set to STREAMING_RESPONSE
[07:11:37][D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[07:11:37][D][light:036]: 'Echo Assistant - Bathroom a02af0' Setting:
[07:11:37][D][light:047]:   State: OFF
[07:11:37][D][light:036]: 'Echo Assistant - Bathroom a02af0' Setting:
[07:11:37][D][light:047]:   State: ON
[07:11:37][D][light:051]:   Brightness: 100%
[07:11:37][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[07:11:37][D][speaker_media_player:420]: State changed to ANNOUNCING
[07:11:37][D][micro_wake_word:129]: Stopping Microphone
[07:11:37][D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[07:11:37][D][esp-idf:000]: I (185328) I2S: DMA queue destroyed

[07:11:37][D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE
WARNING bathroom-assistant-a02af0 @ 192.168.x.y: Connection error occurred: [Errno 104] Connection reset by peer

```

LED turns blue, then a hard crash/reboot

awaldram, 2025-02-21:
> 
> Edit: I'm noticing that all of the reports, including mine, are using https. I wonder if that's the root of the problem.

I thought similar but it cn't be local setup as I use 'home built' voice assistant running arduino based esphome and re-speaker using IDF 5.2.0 neither have any issue with playing from a tls source point.

My home brew uses mp3 and the respeaker FLAC.

Interestingly the re-speaker uses FLAC due to it low resource usage . probably worthe looking at the re-speaker code from [formatBCE](https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration/commits?author=formatBCE) as it seems very close to the VA developers code base and does work :)

Here is my home brew running the same sequence as my M5 example above (without crashing) , Yes its seriously under resourced but is only used to run some night sequences from saying 'goodnight' so it only needs to manage 'Done'.

[14:22:58][D][voice_assistant:642]: Event Type: 8
[14:22:58][D][voice_assistant:720]: Response URL: "https://xxxxxx.duckdns.org:8123/api/tts_proxy/iaNFoHG--1_BCncmb0PCfw.mp3"
[14:22:58][D][voice_assistant:506]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[14:22:58][D][voice_assistant:512]: Desired state set to STREAMING_RESPONSE
[14:22:58][D][media_player:073]: 'Player' - Setting
[14:22:58][D][media_player:080]:   Media URL: https://xxxxxxx.duckdns.org:8123/api/tts_proxy/iaNFoHG--1_BCncmb0PCfw.mp3
[14:22:58][D][media_player:086]:  Announcement: yes
[14:22:58][D][voice_assistant:642]: Event Type: 2
[14:22:58][D][voice_assistant:736]: Assist Pipeline ended
[14:23:00][W][component:237]: Component i2s_audio.media_player took a long time for an operation (2106 ms).
[14:23:01][W][component:238]: Components should block for at most 30 ms.
[14:23:01][D][voice_assistant:506]: State changed from STREAMING_RESPONSE to IDLE
[14:23:01][D][voice_assistant:512]: Desired state set to IDLE
[14:23:01][D][light:036]: 'Light' Setting:
[14:23:01][D][light:047]:   State: OFF
[14:23:01][D][light:085]:   Transition length: 1.0s
[14:23:03][W][component:237]: Component i2s_audio.media_player took a long time for an operation (371 ms).
[14:23:03][W][component:238]: Components should block for at most 30 ms.
[14:23:03][W][component:237]: Component i2s_audio.media_player took a long time for an operation (368 ms).
[14:23:03][W][component:238]: Components should block for at most 30 ms.

righteouskill100, 2025-02-22:
I am seeing the same thing after the latest update. I can get one response out of it before it crashes and resets itself. any idea how to roll the firmware back?

awaldram, 2025-02-23:
> I am seeing the same thing after the latest update. I can get one response out of it before it crashes and resets itself. any idea how to roll the firmware back?

The original project should still be the old build (was yesterday). So restart from the project page.

If not, then  in the closed issue m5 won't compile I changed the code to use the speaker rather than media player.

Drop that into you m5 , make any changes needed to localise ( API keys, WiFi etc) and compile.

neurove, 2025-02-23:
I can confirm that the issue only occurs when using https.
My HA setup uses lets encrypt certificates and the M5 Echo reboots on announce.
When I remove the ssl certificates and update the network settings to use http, then announce and the setup wizard works normally.

tescophil, 2025-02-24:
Same this for me, crash on announce

HarlemSquirrel, 2025-02-26:
Same issue after updating to 2025.2.1

rubin110, 2025-02-26:
I should have noted, my cert comes from Let's Encrypt. I tried the M5 Atom Echo media player build from the ready-made projects page, and the device will play WAV files from HA over SSL just fine.

HarlemSquirrel, 2025-02-26:
I am also using Let's Encrypt

Taomyn, 2025-02-26:
So for my situation it means that either I need a way to add my CA certificate to the store used so that it's trusted, or for the "verify_ssl: false" to be available so that it ignores the problem. Perhaps then it will be able to read the sound file but then crash out like yours are doing.

HarlemSquirrel, 2025-02-26:
I think you can add a section below where the package is imported to override configs 

https://esphome.io/components/http_request.html

Something like this

```yml
http_request:
  verify_ssl: false
```

Taomyn, 2025-02-26:
> Something like this
> 
> http_request:
>   verify_ssl: false

I tried exactly that with 2025.2.0 and it would not compile, hence why I submitted the issue, but I now tried it again as I wanted to grab the complier error for putting here and would you believe it, the compile worked.

I'm away from device at the moment so I can test actual speech, but I will do so later and report if mine now crashes like is being reported.

cineafx, 2025-02-26:
> http_request:
>   verify_ssl: false

This seems to fix the issue for me. I just added it at the end of the yaml file and recompiled it. (Using version 2025.2.1 of esphome.) 
Thanks!

Taomyn, 2025-02-26:
Unfortunately not for me, I still get:

```
[18:29:46][D][voice_assistant:675]: Intent started
[18:29:46][D][voice_assistant:642]: Event Type: 6
[18:29:46][D][voice_assistant:642]: Event Type: 7
[18:29:46][D][voice_assistant:698]: Response: "Turned off the light"
[18:29:46][D][light:036]: 'M5Stack Atom Echo 31' Setting:
[18:29:46][D][light:051]:   Brightness: 100%
[18:29:46][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[18:29:46][D][light:109]:   Effect: 'None'
[18:29:46][D][voice_assistant:642]: Event Type: 8
[18:29:46][D][voice_assistant:720]: Response URL: "https://hass.mydomain.com/api/tts_proxy/WLOXHKib_IBW_oqYMXtc7g.wav"
[18:29:46][D][voice_assistant:506]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[18:29:46][D][voice_assistant:512]: Desired state set to STREAMING_RESPONSE
[18:29:46][D][media_player:073]: 'M5Stack Atom Echo 31' - Setting
[18:29:46][D][media_player:080]:   Media URL: https://hass.mydomain.com/api/tts_proxy/WLOXHKib_IBW_oqYMXtc7g.wav
[18:29:46][D][media_player:086]:  Announcement: yes
[18:29:46][D][light:036]: 'M5Stack Atom Echo 31' Setting:
[18:29:46][D][light:051]:   Brightness: 100%
[18:29:46][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[18:29:46][D][speaker_media_player:420]: State changed to ANNOUNCING
[18:29:46][D][voice_assistant:642]: Event Type: 2
[18:29:46][D][voice_assistant:736]: Assist Pipeline ended
[18:29:46][D][esp-idf:000][ann_read]: E (21422) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780

[18:29:46][D][esp-idf:000][ann_read]: E (21429) esp-tls: Failed to open new connection

[18:29:46][D][esp-idf:000][ann_read]: E (21435) TRANSPORT_BASE: Failed to open a new connection

[18:29:46][D][esp-idf:000][ann_read]: E (21445) HTTP_CLIENT: Connection failed, sock < 0

[18:29:46][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
```

I made sure to clean the build files beforehand just to be sure, and confirmed the WAV could be played from another machine.

[logs_m5ae-31_run.txt](https://github.com/user-attachments/files/18989738/logs_m5ae-31_run.txt)
[m5ae-31.yaml.txt](https://github.com/user-attachments/files/18989739/m5ae-31.yaml.txt)

josephperry, 2025-02-26:
> > http_request:
> > verify_ssl: false
> 
> This seems to fix the issue for me. I just added it at the end of the yaml file and recompiled it. (Using version 2025.2.1 of esphome.) Thanks!

This also fixed it for me.  It wasn't crashing, but there was no voice output until I made this addition.

Taomyn, 2025-02-26:
@cineafx @josephperry could you share your yaml file so I can compare it to mine? I must have something else that is blocking the command or it simply doesn't work even with my certificate.

cineafx, 2025-02-26:
> [@cineafx](https://github.com/cineafx) [@josephperry](https://github.com/josephperry) could you share your yaml file so I can compare it to mine? I must have something else that is blocking the command or it simply doesn't work even with my certificate.

Sure:
```yaml
substitutions:
  name: m5stack-atom-echo-fde114
  friendly_name: M5Stack Atom Echo fde114
packages:
  m5stack.atom-echo-wake-word-voice-assistant: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: XXXXXXXX

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

http_request:
  verify_ssl: false
```

Taomyn, 2025-02-26:
Thanks @cineafx pretty much what I have, so it got me thinking about anything else that has recently changed. I recalled updating my proxy server but that made no sense because internally I should be talking directly to my HA server. Appears I'm not, and the change to the proxy included fixing SSL web connections to TLS1.3 - so for the internal proxy I set it to TLS1.2+ and bang, it started working.

So add another limitation: seems it does not support TLS1.3

No crashing but I do now hear the voice response, but it's extremely low volume and barely audible. It was perfectly loud before all the problems, so something else I need to look into now.

cineafx, 2025-02-26:
> but it's extremely low volume and barely audible. It was perfectly loud before all the problems

The new media player has a volume entity:
[image]

Taomyn, 2025-02-26:
@cineafx you're a superstar, had not spotted that and works a treat. I was recalling this happening when I first got the M5AE last year and was going through old files to see if I had added anything to the yaml at the time.

brettmiller, 2025-02-27:
I was also  able to get it working with:
```
http_request:
  verify_ssl: false
``` 
Though, it seems odd because I've got a valid Let's Encrypt TLS cert, so it should verify.  My reverse proxy that hosts the cert supports TLSv1.2, so if v1.3 is also an issue I luckily didn't hit that.

HarlemSquirrel, 2025-03-03:
Yeah same here. Even though I have Let's Encrypt certificates, turning off SSL verify worked. This wasn't a problem for me until a recent update.

shtrom, 2025-03-07:
Another confirmation that disabling TLS verification works as a stopgap with a Let's Encrypt cert.

tescophil, 2025-03-07:
> I think you can add a section below where the package is imported to override configs
> 
> https://esphome.io/components/http_request.html
> 
> Something like this
> 
> http_request:
>   verify_ssl: false

I have this problem on home assistant, forgive my ignorance, but where / which file is this added to exactly? (all my config is via the UI)

shtrom, 2025-03-07:
> I have this problem on home assistant, forgive my ignorance, but where / which file is this added to exactly? (all my config is via the UI)

Are you running the ESPHome dashboard locally?

[image]

You should `Edit` the YAML file, add that at the bottom, and `Install` again.

I don't think you can change it on the pre-built binaries.

tescophil, 2025-03-07:
No, just via home assistant...., guessed as much,.,, will have to wait for an actual fix to come out.

Thanks.

bigboban, 2025-03-15:
For all with SSL/DNS/HTTP_CLIENT problem - i SOLVED this here: https://github.com/esphome/issues/issues/6807.

teekennedy, 2025-04-11:
I got a Home Assistant Voice Preview Edition recently and noticed it doesn't have any TLS related errors. Digging into the HAVPE ESPHome config I noticed [this line](https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml#L66) which is highly relevant to the errors we're seeing.

If TLS 1.3 is not enabled by default on esphome that's sure to cause problems. I don't have the time to test this myself right now, but I think if you add

```yaml
esp32:
  framework:
    sdkconfig_options:
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "y"
```

to your configuration, it should work without the need to set `verify_ssl: false`.

brettmiller, 2025-04-11:
I test the `CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "y"` setting doesn't resolve the issue. It seems like it maybe makes it a little less prone to crashing, but it never plays the announcement audio.

carrot016, 2025-04-20:
Added this
> ```
> http_request:
>   verify_ssl: false
> ```
But still does not work for me

closed 2025-05-27 by github-actions[bot]