esphome/wake-word-voice-assistants #72: Error mediaplayer after upgrading esp32-s3-box-3 with esphome 2025.2

ericvb, 2025-02-22:
With the previous version, no problem in voice responses on questions/command.
This is my yaml file.
```
substitutions:
  name: esp32-s3-box-3-054618
  friendly_name: ESP32 S3 Box 3
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: mysecret1

ota:
  - platform: esphome
    password: mysecret2
    
wifi:
  ssid: myssid
  password: mysecret3
```

After compiling the new version with esphome 2025.2 with the media player component, I got no voice response anymore.
In the logfile I see an error in the announcement pipeline.
```
[20:16:44][D][micro_wake_word:160]: The 'Okay Nabu' model sliding average probability is 0.975 and most recent probability is 1.000
[20:16:44][D][micro_wake_word:123]: Wake Word 'Okay Nabu' Detected
[20:16:44][D][micro_wake_word:194]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[20:16:44][D][micro_wake_word:129]: Stopping Microphone
[20:16:44][D][micro_wake_word:194]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[20:16:44][D][micro_wake_word:194]: State changed from STOPPING_MICROPHONE to IDLE
[20:16:44][D][voice_assistant:505]: State changed from IDLE to START_MICROPHONE
[20:16:44][D][voice_assistant:512]: Desired state set to START_PIPELINE
[20:16:44][D][voice_assistant:223]: Starting Microphone
[20:16:44][D][voice_assistant:505]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[20:16:44][D][voice_assistant:505]: State changed from STARTING_MICROPHONE to START_PIPELINE
[20:16:44][D][voice_assistant:277]: Requesting start...
[20:16:44][D][voice_assistant:505]: State changed from START_PIPELINE to STARTING_PIPELINE
[20:16:44][D][voice_assistant:527]: Client started, streaming microphone
[20:16:44][D][voice_assistant:505]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[20:16:44][D][voice_assistant:512]: Desired state set to STREAMING_MICROPHONE
[20:16:44][D][voice_assistant:642]: Event Type: 1
[20:16:44][D][voice_assistant:645]: Assist Pipeline running
[20:16:44][D][voice_assistant:642]: Event Type: 3
[20:16:44][D][voice_assistant:656]: STT started
[20:16:44][D][text_sensor:064]: 'text_request': Sending state '...'
[20:16:44][D][text_sensor:064]: 'text_response': Sending state '...'
[20:16:44][W][component:237]: Component voice_assistant took a long time for an operation (306 ms).
[20:16:44][W][component:238]: Components should block for at most 30 ms.
[20:16:45][D][voice_assistant:642]: Event Type: 11
[20:16:45][D][voice_assistant:805]: Starting STT by VAD
[20:16:50][D][voice_assistant:642]: Event Type: 12
[20:16:50][D][voice_assistant:809]: STT by VAD end
[20:16:50][D][voice_assistant:505]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[20:16:50][D][voice_assistant:512]: Desired state set to AWAITING_RESPONSE
[20:16:50][D][voice_assistant:505]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[20:16:50][W][component:237]: Component voice_assistant took a long time for an operation (321 ms).
[20:16:50][W][component:238]: Components should block for at most 30 ms.
[20:16:50][D][voice_assistant:505]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[20:16:50][D][voice_assistant:642]: Event Type: 4
[20:16:50][D][voice_assistant:670]: Speech recognised as: "wat voor weer is het nu"
[20:16:50][D][text_sensor:064]: 'text_request': Sending state 'wat voor weer is het nu'
[20:16:51][W][component:237]: Component voice_assistant took a long time for an operation (324 ms).
[20:16:51][W][component:238]: Components should block for at most 30 ms.
[20:16:51][D][voice_assistant:642]: Event Type: 5
[20:16:51][D][voice_assistant:675]: Intent started
[20:16:51][D][voice_assistant:642]: Event Type: 6
[20:16:51][D][voice_assistant:642]: Event Type: 7
[20:16:51][D][voice_assistant:698]: Response: "10.0 graden met regen"
[20:16:51][D][text_sensor:064]: 'text_response': Sending state '10.0 graden met regen'
[20:16:51][W][component:237]: Component voice_assistant took a long time for an operation (345 ms).
[20:16:51][W][component:238]: Components should block for at most 30 ms.
[20:16:51][D][voice_assistant:642]: Event Type: 8
[20:16:51][D][voice_assistant:720]: Response URL: "http://192.168.1.25:8123/api/tts_proxy/I0i_sgTOnnWGLS_q1D-Fmw.flac"
[20:16:51][D][voice_assistant:505]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[20:16:51][D][voice_assistant:512]: Desired state set to STREAMING_RESPONSE
[20:16:51][D][media_player:073]: 'ESP32 S3 Box 3 Living' - Setting
[20:16:51][D][media_player:080]:   Media URL: http://192.168.1.25:8123/api/tts_proxy/I0i_sgTOnnWGLS_q1D-Fmw.flac
[20:16:51][D][media_player:086]:  Announcement: yes
[20:16:51][D][speaker_media_player:420]: State changed to ANNOUNCING
[20:16:51][D][voice_assistant:642]: Event Type: 2
[20:16:51][D][voice_assistant:736]: Assist Pipeline ended
[20:16:51][E][speaker_media_player:334]: The announcement pipeline's file reader encountered an error.
[20:16:56][D][esp-idf:000][ann_read]: E (73818675) esp-tls: [sock=58] select() timeout

[20:16:56][D][esp-idf:000][ann_read]: E (73818679) transport_base: Failed to open a new connection: 32774

[20:16:56][D][esp-idf:000][ann_read]: E (73818682) HTTP_CLIENT: Connection failed, sock < 0

[20:16:56][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
[20:16:56][D][micro_wake_word:399]: Resetting buffers and probabilities
[20:16:56][D][micro_wake_word:194]: State changed from IDLE to START_MICROPHONE
[20:16:56][D][speaker_media_player:420]: State changed to IDLE
[20:16:56][W][component:237]: Component speaker.media_player took a long time for an operation (324 ms).
[20:16:56][W][component:238]: Components should block for at most 30 ms.
[20:16:56][D][micro_wake_word:107]: Starting Microphone
[20:16:56][D][micro_wake_word:194]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[20:16:56][D][voice_assistant:505]: State changed from STREAMING_RESPONSE to IDLE
[20:16:56][D][voice_assistant:512]: Desired state set to IDLE
[20:16:57][W][micro_wake_word:173]: Wake word is already running
[20:16:57][W][component:237]: Component voice_assistant took a long time for an operation (307 ms).
[20:16:57][W][component:238]: Components should block for at most 30 ms.
[20:16:57][D][micro_wake_word:194]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
```

Any idea about the problem?

I've also a m5stack-atom running as voice assistant. Compiled also a new version for it with the esphome 2025.2 version.
Is working without any problem.

casgh23, 2025-02-23:
I had the same problem but noticed that its trying to connect over https instead of http on port 8123. Turned out to be an incorrectly configured "local url" in the homeassistant network settings.

casgh23, 2025-02-24:
This not in the yaml config its in the gui in home assistant. Go to settings -> system -> network and check the "local url" make sure it is correct and reachable by your voice satelite. that fixed it for me

ericvb, 2025-02-24:
Verified, it is correct.

[image]

When I test with the Copy link result in the browser I can access my HA.

ericvb, 2025-02-26:
Today I recompiled the whole script with EspHome 2025.2.1, but still same problem :-(

bobzer, 2025-03-08:
Samething here but also for the m5stack, was perfectly working before the update and not anymore.
no voice answer anymore, no audio what's so ever, 
When I try the media player I see the screen flash like something happens but that's it.

ericvb, 2025-03-23:
Today I installed the new version of ESPHome 2025.3.1 and recompiled the same from above, ESP32 S3 Box 3 yaml file.
Everything is working again, I get an audio response when I ask questions or give commands!

closed 2025-03-23 by ericvb

carrot016, 2025-04-20:
> This not in the yaml config its in the gui in home assistant. Go to settings -> system -> network and check the "local url" make sure it is correct and reachable by your voice satelite. that fixed it for me

Worked in the way that I don't have the error log anymore, but I still don't have sound.