esphome/wake-word-voice-assistants #74: In home assistant wake word stops working after first use in m5stack-atom-echo.yaml

keyboarderror, 2025-02-24:
Using in home assistant wake words on the Atom Echo causes the device to hang after the first request. On device works normally. You can reload the integration or restart the device to use it again.

I've fixed the problem by the changing the else condition in the start_wake_word script like so:

Before:

```
- id: start_wake_word
    then:
      - wait_until:
          and:
            - media_player.is_idle:
            - speaker.is_stopped:
      - if:
          condition:
            lambda: return id(wake_word_engine_location).state == "On device";
          then:
            - voice_assistant.stop
            - micro_wake_word.stop:
            - delay: 1s
            - script.execute: reset_led
            - script.wait: reset_led
            - micro_wake_word.start:
          else:
            - if:
                condition: voice_assistant.is_running
                then:
                  - voice_assistant.stop:
                  - script.execute: reset_led
            - voice_assistant.start_continuous:`

After:

````- id: start_wake_word
    then:
      - wait_until:
          and:
            - media_player.is_idle:
            - speaker.is_stopped:
      - if:
          condition:
            lambda: return id(wake_word_engine_location).state == "On device";
          then:
            - voice_assistant.stop
            - micro_wake_word.stop:
            - delay: 1s
            - script.execute: reset_led
            - script.wait: reset_led
            - micro_wake_word.start:
          else:
            - if:
                condition: 
                  lambda: return id(wake_word_engine_location).state == "In Home Assistant";
                then:
                  - voice_assistant.stop:
                  - script.execute: reset_led
                  - micro_wake_word.stop
                  - delay: 500ms
                  - lambda: id(va).set_use_wake_word(true);               
            - voice_assistant.start_continuous

But I really have no idea what I'm doing. You might want to fix it properly.

Androxilogin, 2025-02-26:
I've noticed that pressing the button while using home assistant wake words will enable it to listen to a new request. Since the last update it has been working as an on/off switch, so it seems.

jesserockz, 2025-02-27:
I am unable to reproduce this. I reflashed the latest released firmware just now and it is working and listening for a wake word after every response given. I have tried over 20 times so far.

Can you provide device logs showing what you mean by stops working?

keyboarderror, 2025-02-27:
OK, I changed the script back and re-flashed it. I compiled the latest firmware last night. It takes an hour and 10 minutes on my Raspberry Pi 4. The problem is back. Here's the log from re-flash to fail. 

[logs_m5stack-atom-echo_run.txt](https://github.com/user-attachments/files/19000369/logs_m5stack-atom-echo_run.txt)

bilbotbaggin, 2025-02-27:
I Can confirm the same behavior on two M5 Atom Echo devices cleanly installed with the voice assist firmware, I am able to issue one voice command and then the device becomes non responsive to voice commands.

Nothing jumping out to me on the device console though I am fairly inexperienced.

keyboarderror, 2025-02-27:
My workaround is just copied from the select: -platform: template that does the initial setup. The - micro_wake_word.stop statement isn't necessary looking at the logs as it's part on the On device wake word service and is set at start.

jesserockz, 2025-02-27:
@keyboarderror looking at those logs, you are not even using micro-wake-word, but it is just always streaming to ha which is doing the wake word detection with openwakeword etc?

keyboarderror, 2025-02-27:
@jesserockz Correct. It was an unnecessary command leftover from the copied code.

closed 2025-05-08 by github-actions[bot]