esphome/wake-word-voice-assistants #76: Compilation fails or reboots HA

AbidingOhmsLaw, 2025-02-25:
I'm having trouble compiling the esp-S3-box yaml after the 2025.2.0 update. If I compile inside using the HA Settings\Updates screen then it runs for a bit and HA then crashes and reboots.  If I compile using the ESPHome add-in I get the error below:

My HA instance is up to date and is running on a Raspberry 4 with HA OS
```
Home Assistant
Core     2025.2.5
Supervisor    2025.02.1
Operating System    14.2
Frontend    20250221.0
Hardware   Raspberry Pi 4 rpi4-64
```

I have tried both using "compile_process_limit: 1" and without it and get the same results

My yaml is copied directly from this git with the only changes being the these lines
```
substitutions:
  name: esp32-s3-box-3-5ac17c
  friendly_name: ESP32 S3 Box 3 5ac17c
.
.
.

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2025.2.0
  # compile_process_limit: 1  (tried this line commented out and not)
  name_add_mac_suffix: true
  on_boot:
    priority: 600
    then:
      - script.execute: draw_display
      - delay: 30s
      - if:
          condition:
            lambda: return id(init_in_progress);
          then:
            - lambda: id(init_in_progress) = false;
            - script.execute: draw_display
.
.
.

api:
  on_client_connected:
    - script.execute: draw_display
  on_client_disconnected:
    - script.execute: draw_display
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx (redacted)
.
.
.
wifi:
  ap:
  on_connect:
    - script.execute: draw_display
  on_disconnect:
    - script.execute: draw_display
  ssid: !secret wifi_ssid
  password: !secret wifi_password
.
.
.
```

Below is the error I get when compiling in the ESPHome add-in

```
Compiling .pioenvs/esp32-s3-box-3-5ac17c/src/main.cpp.o
In file included from src/esphome/components/i2s_audio/i2s_audio.h:5,
                 from src/esphome.h:51,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
In file included from components/esp-tflite-micro/tensorflow/lite/micro/tflite_bridge/flatbuffer_conversions_bridge.h:19,
                 from components/esp-tflite-micro/tensorflow/lite/micro/micro_allocator.h:26,
                 from components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter.h:26,
                 from src/esphome/components/micro_wake_word/streaming_model.h:8,
                 from src/esphome/components/micro_wake_word/micro_wake_word.h:6,
                 from src/esphome.h:83:
components/esp-tflite-micro/tensorflow/lite/core/api/flatbuffer_conversions.h: In member function 'T* tflite::BuiltinDataAllocator::AllocatePOD()':
components/esp-tflite-micro/tensorflow/lite/core/api/flatbuffer_conversions.h:46:24: warning: 'template<class _Tp> struct std::is_pod' is deprecated: use is_standard_layout && is_trivial instead [-Wdeprecated-declarations]
   46 |     static_assert(std::is_pod<T>::value, "Builtin data structure must be POD.");
      |                        ^~~~~~
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/bits/stl_pair.h:60,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/bits/stl_algobase.h:64,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/deque:60,
                 from src/esphome/components/api/api_frame_helper.h:3,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome.h:3:
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/type_traits:757:5: note: declared here
  757 |     is_pod
      |     ^~~~~~
xtensa-esp32s3-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/esp32-s3-box-3-5ac17c/src/main.cpp.o] Error 1
========================= [FAILED] Took 96.39 seconds ========================= 
```

ericvb, 2025-02-26:
I see that you tried already with the option "compile_process_limit: 1" but without any luck.

Your HA needs more memory to be able to compile the whole thing.
I had running HA in a VM (virtual machine) with 3 Gb of RAM, but I had to add 1 Gb RAM extra to be able to compile successfully.

Try to disable other add-ins to free memory on your PI.

jesserockz, 2025-02-27:
Unfortunately this is the downsides of using these low powered boards for Home Assistant. While they can run the (Home Assistant) software perfectly well and some add-ons etc, actually **compiling** software or firmware is an entirely different thing and requires a lot of resource.

AbidingOhmsLaw, 2025-03-01:
OK, This is mostly solved, or at least I have a different issue now.  I downloaded ESPHome to my laptop and compiled the yaml copied from the ESPHome on the Pi 4 on the laptop.  I used the web tool to upload the firmware to the s3 box.  It works mostly there seems to be some issue with it staying connected to HA and the ESPHome instance on the Pi, it's connecting when booted but then goes "offline" after about 5 min.

jesserockz, 2025-03-19:
Please use a different issue for different problems

closed 2025-03-19 by jesserockz