esphome/wake-word-voice-assistants #79: Problem with restart ESP32-S3-BOX-3 using manual trigger button

pschmidtpl, 2025-03-05:
I have an ESP32-S3-BOX-3 integrated with Home Assistant.
I want to create a button that wakes it up and switches it to listening mode without a wake-up word.
I added a button

```
button:
  - platform: template
    name: Wakeword
    id: wakeword_trigger_external
    on_press: 
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);
      - delay: 500ms
      - voice_assistant.start:
          wake_word: !lambda return "okay nabu";
      - delay: 5000ms
      - lambda: id(va).set_use_wake_word(true);  
```

Unfortunately, triggering it causes the device to restart when recognizing a command.
I have also try simple:

```
      - voice_assistant.start:
          wake_word: !lambda return "okay nabu";
```

with the same result (restat device)

When waking up via the wake word, everything works correctly (logs as follow):

_[14:05:11][C][api:141]:   Address: esp32-s3-box-3-05b950.local:6053
[14:05:11][C][api:143]:   Using noise encryption: YES
[14:05:11][C][micro_wake_word:051]: microWakeWord:
[14:05:11][C][micro_wake_word:052]:   models:
[14:05:11][C][micro_wake_word:015]:     - Wake Word: Okay Nabu
[14:05:11][C][micro_wake_word:016]:       Probability cutoff: 0.970
[14:05:11][C][micro_wake_word:017]:       Sliding window size: 5
[14:05:13][D][micro_wake_word:160]: The 'Okay Nabu' model sliding average probability is 0.987 and most recent probability is 1.000
[14:05:13][D][micro_wake_word:123]: Wake Word 'Okay Nabu' Detected
[14:05:13][D][micro_wake_word:194]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[14:05:13][D][micro_wake_word:129]: Stopping Microphone
[14:05:13][D][micro_wake_word:194]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[14:05:13][D][micro_wake_word:194]: State changed from STOPPING_MICROPHONE to IDLE
[14:05:13][D][voice_assistant:505]: State changed from IDLE to START_MICROPHONE
[14:05:13][D][voice_assistant:512]: Desired state set to START_PIPELINE
[14:05:14][D][voice_assistant:223]: Starting Microphone
[14:05:14][D][voice_assistant:505]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[14:05:14][D][voice_assistant:505]: State changed from STARTING_MICROPHONE to START_PIPELINE
[14:05:14][D][voice_assistant:277]: Requesting start...
[14:05:14][D][voice_assistant:505]: State changed from START_PIPELINE to STARTING_PIPELINE
[14:05:14][D][voice_assistant:527]: Client started, streaming microphone
[14:05:14][D][voice_assistant:505]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[14:05:14][D][voice_assistant:512]: Desired state set to STREAMING_MICROPHONE
[14:05:14][D][voice_assistant:642]: Event Type: 1
[14:05:14][D][voice_assistant:645]: Assist Pipeline running
[14:05:14][D][voice_assistant:642]: Event Type: 3
[14:05:14][D][voice_assistant:656]: STT started
[14:05:14][D][text_sensor:064]: 'text_request': Sending state '...'
[14:05:14][D][text_sensor:064]: 'text_response': Sending state '...'
[14:05:14][W][component:237]: Component voice_assistant took a long time for an operation (229 ms).
[14:05:14][W][component:238]: Components should block for at most 30 ms.
[14:05:16][D][voice_assistant:642]: Event Type: 11
[14:05:16][D][voice_assistant:805]: Starting STT by VAD
[14:05:19][D][voice_assistant:642]: Event Type: 12
[14:05:19][D][voice_assistant:809]: STT by VAD end
[14:05:19][D][voice_assistant:505]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[14:05:19][D][voice_assistant:512]: Desired state set to AWAITING_RESPONSE
[14:05:19][D][voice_assistant:505]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE_

Here are the logs of the incorrect operation of the device (device restart) when woken up by the button. The device wakes up correctly but restarts when attempting to recognize a command:

_[14:06:03][C][micro_wake_word:051]: microWakeWord:
[14:06:03][C][micro_wake_word:052]:   models:
[14:06:03][C][micro_wake_word:015]:     - Wake Word: Okay Nabu
[14:06:03][C][micro_wake_word:016]:       Probability cutoff: 0.970
[14:06:03][C][micro_wake_word:017]:       Sliding window size: 5
[14:06:07][D][button:010]: 'Wakeword' Pressed.
[14:06:07][D][voice_assistant:505]: State changed from IDLE to START_MICROPHONE
[14:06:07][D][voice_assistant:512]: Desired state set to START_PIPELINE
[14:06:07][D][voice_assistant:223]: Starting Microphone
[14:06:07][D][voice_assistant:505]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[14:06:07][D][voice_assistant:505]: State changed from STARTING_MICROPHONE to START_PIPELINE
[14:06:07][D][voice_assistant:277]: Requesting start...
[14:06:07][D][voice_assistant:505]: State changed from START_PIPELINE to STARTING_PIPELINE
[14:06:07][D][voice_assistant:527]: Client started, streaming microphone
[14:06:07][D][voice_assistant:505]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[14:06:07][D][voice_assistant:512]: Desired state set to STREAMING_MICROPHONE
[14:06:07][D][voice_assistant:642]: Event Type: 1
[14:06:07][D][voice_assistant:645]: Assist Pipeline running
[14:06:07][D][voice_assistant:642]: Event Type: 3
[14:06:07][D][voice_assistant:656]: STT started
[14:06:07][D][text_sensor:064]: 'text_request': Sending state '...'
[14:06:07][D][text_sensor:064]: 'text_response': Sending state '...'
[14:06:07][W][component:237]: Component voice_assistant took a long time for an operation (227 ms).
[14:06:07][W][component:238]: Components should block for at most 30 ms.
[14:06:10][D][voice_assistant:642]: Event Type: 11
[14:06:10][D][voice_assistant:805]: Starting STT by VAD
[14:06:12][D][voice_assistant:642]: Event Type: 12
[14:06:12][D][voice_assistant:809]: STT by VAD end
[14:06:12][D][voice_assistant:505]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[14:06:12][D][voice_assistant:512]: Desired state set to AWAITING_RESPONSE
[14:06:12][D][voice_assistant:505]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[14:06:12][W][component:237]: Component voice_assistant took a long time for an operation (238 ms).
[14:06:12][W][component:238]: Components should block for at most 30 ms.
[14:06:12][D][voice_assistant:642]: Event Type: 4_

This is where the reset occurs.

Full file esp32-s3-box-3-05b950.yaml:

```
substitutions:
  name: esp32-s3-box-3-05b950
  friendly_name: ESP32 S3 Box 3 05b950

  loading_illustration_file: https://github.com/jlpouffier/home-assistant-s3-box-community-illustrations/raw/main/ava_midjourney/illustrations/loading.png
  idle_illustration_file: https://github.com/jlpouffier/home-assistant-s3-box-community-illustrations/raw/main/ava_midjourney/illustrations/loading.png
  listening_illustration_file: https://github.com/jlpouffier/home-assistant-s3-box-community-illustrations/raw/main/ava_midjourney/illustrations/idle.png
  thinking_illustration_file: https://github.com/jlpouffier/home-assistant-s3-box-community-illustrations/raw/main/ava_midjourney/illustrations/thinking.png
  replying_illustration_file: https://github.com/jlpouffier/home-assistant-s3-box-community-illustrations/raw/main/ava_midjourney/illustrations/replying.png
  error_illustration_file: https://github.com/jlpouffier/home-assistant-s3-box-community-illustrations/raw/main/ava_midjourney/illustrations/error.png
  

packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: tHnTyvkq0lYl1NQR0bv+BCBPaw/fAkk2EgZ4Wvxv7jE=
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

text_sensor:
  - platform: template
    id: text_user_request
    name: "Text Request"
    internal: false
  - platform: template
    id: text_agent_response
    name: "Text Response"
    internal: false

voice_assistant:
  on_stt_end:
    - text_sensor.template.publish:
        id: text_user_request
        state: !lambda return x;
  on_tts_start:
    - text_sensor.template.publish:
        id: text_agent_response
        state: !lambda return x;

button:
  - platform: template
    name: Wakeword
    id: wakeword_trigger_external
    on_press: 
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);
      - delay: 500ms
      - voice_assistant.start:
          wake_word: !lambda return "okay nabu";
      - delay: 5000ms
      - lambda: id(va).set_use_wake_word(true);
```

BigBobbas, 2025-03-06:
just start voice assistant, you don't need to return a wakeword as you are bypassing that part

pschmidtpl, 2025-03-08:
```
button:
  - platform: template
    name: Wakeword
    id: wakeword_trigger_external
    on_press: 
      - voice_assistant.start:
```

Unfortunately, the same result — a restart during speech recognition

mitrokun, 2025-03-11:
A few seconds longer, but works smoothly
```
button:
  - platform: restart
    name: restart
```

closed 2025-05-07 by github-actions[bot]