esphome/wake-word-voice-assistants #80: ESP32-s3-box-3 crash when using Click to Converse

christianaubry, 2025-03-06:
I tried to implement the Click to Converse example but got ESP32 crash right after speech recognition. Note that speech recognition seems to times out. From Home assistant point of view process goes to its end including streaming appropriate response as Satellite Assist goes to respond state.

ESP Log
[23:00:40]I (609) esp_image: segment 1: paddr=0032d2e8 vaddr=3fc9ab[I][logger:171]: Log initialized
[23:00:40][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[23:00:40][D][esp32.preferences:114]: Saving 1 preferences to flash...
[23:00:40][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[23:00:40][I][app:029]: Running through setup()...
[23:00:40][C][i2c.idf:017]: Setting up I2C bus...
[23:00:40][I][i2c.idf:256]: Performing I2C bus recovery
[23:00:47][D][esp-idf:000]: I (1529) gp[D][switch:012]: 'restart conversation' Turning ON.
[23:00:47][D][switch:055]: 'restart conversation': Sending state ON
[23:00:47][D][binary_sensor:036]: 'continuous_conversation': Sending state ON
[23:00:47][D][voice_assistant:505]: State changed from IDLE to START_MICROPHONE
[23:00:47][D][voice_assistant:512]: Desired state set to START_PIPELINE
[23:00:47][D][voice_assistant:223]: Starting Microphone
[23:00:47][D][ring_buffer:034]: Created ring buffer with size 16384
[23:00:47][D][voice_assistant:505]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[23:00:47][D][voice_assistant:505]: State changed from STARTING_MICROPHONE to START_PIPELINE
[23:00:47][D][voice_assistant:277]: Requesting start...
[23:00:47][D][voice_assistant:505]: State changed from START_PIPELINE to STARTING_PIPELINE
[23:00:47][D][voice_assistant:527]: Client started, streaming microphone
[23:00:47][D][voice_assistant:505]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[23:00:47][D][voice_assistant:512]: Desired state set to STREAMING_MICROPHONE
[23:00:47][D][voice_assistant:642]: Event Type: 1
[23:00:47][D][voice_assistant:645]: Assist Pipeline running
[23:00:47][D][voice_assistant:642]: Event Type: 3
[23:00:47][D][voice_assistant:656]: STT started
[23:00:47][D][text_sensor:064]: 'text_request': Sending state '...'
[23:00:47][D][text_sensor:064]: 'text_response': Sending state '...'
[23:00:48][W][component:237]: Component voice_assistant took a long time for an operation (313 ms).
[23:00:48][W][component:238]: Components should block for at most 30 ms.
[23:00:48][D][switch:016]: 'restart conversation' Turning OFF.
[23:00:48][D][switch:055]: 'restart conversation': Sending state OFF
[23:00:48][D][binary_sensor:036]: 'continuous_conversation': Sending state OFF
[23:00:55][D][voice_assistant:642]: Event Type: 11
[23:00:55][D][voice_assistant:805]: Starting STT by VAD
[23:00:57][D][voice_assistant:642]: Event Type: 12
[23:00:57][D][voice_assistant:809]: STT by VAD end
[23:00:57][D][voice_assistant:505]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[23:00:57][D][voice_assistant:512]: Desired state set to AWAITING_RESPONSE
[23:00:57][D][voice_assistant:505]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[23:00:57][W][component:237]: Component voice_assistant took a long time for an operation (325 ms).
[23:00:57][W][component:238]: Components should block for at most 30 ms.
[23:00:57][D][voice_assistant:642]: Event Type: 4
[23:00:57][D][voice_assistant:670]: Speech recognised as: "Quelle heure est-il ?"
[23:00:57]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[23:00:57]
[23:00:57]Core  0 register dump:
[23:00:57]PC      : 0x42060560  PS      : 0x00060730  A0      : 0x8200f993  A1      : 0x3fcad680  
[23:00:57]A2      : 0x00000000  A3      : 0x3c46f168  A4      : 0x00000200  A5      : 0x3fcad6c0  
[23:00:57]A6      : 0x00000000  A7      : 0x3fca03ac  A8      : 0x3fca03ac  A9      : 0x3fcad670  
[23:00:57]A10     : 0x3fca0920  A11     : 0x00060720  A12     : 0x3fcc4a40  A13     : 0x00060723  
[23:00:57]A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[23:00:57]EXCVADDR: 0x00000014  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
[23:00:57]
[23:00:57]
[23:00:57]Backtrace: 0x4206055d:0x3fcad680 0x4200f990:0x3fcad6c0 0x420156aa:0x3fcad6f0 0x4201600b:0x3fcad710 0x420f468d:0x3fcad790 0x420f4737:0x3fcad7b0 0x42020ed0:0x3fcad7d0 0x42024672:0x3fcad800 0x4200dae2:0x3fcad820
[23:00:57]
[23:00:57]
[23:00:57]
[23:00:57]
[23:00:57]ELF file SHA256: 76a74d4c791fd937
[23:00:57]
[23:00:57]Rebooting...
[23:00:57]ESP-ROM:esp32s3-20210327
[23:00:57]Build:Mar 27 2021
[23:00:57]rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
[23:00:57]Saved PC:0x403777f3
[23:00:57]SPIWP:0xee
[23:00:57]mode:DIO, clock div:1
[23:00:57]load:0x3fce3818,len:0x1750
[23:00:57]load:0x403c9700,len:0x4
[23:00:57]load:0x403c9704,len:0xbe4
[23:00:57]load:0x403cc700,len:0x2d34
[23:00:57]entry 0x403c9908
[23:00:57]I (18) boot: ESP-IDF 5.1.5 2nd stage bootloader
[23:00:57]I (18) boot: compile time Mar  6 2025 22:44:46

HA yaml configuration

substitutions:
  name: jarvis_speaker
  friendly_name: Jarvis Speaker
  micro_wake_word_model: hey_jarvis
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: pdkX9ltbLD7hNleHHGeVnXmFkqQCn8YM/BW2ktzs5TE=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

switch:
  - platform: gpio
    pin: GPIO41
    name: "restart conversation"
    id: restart_conversation

binary_sensor:
  - platform: gpio
    id: continuous_conversation 
    pin: GPIO40
    on_press:
      - if:
          condition: voice_assistant.is_running
          then:
            - voice_assistant.stop:
          else:
            - voice_assistant.start_continuous:

HA version: 2025.3.0

xiasi0, 2025-03-16:
I am experiencing the same issue here: Click to Converse causes the ESP to crash, and MWW cannot work alongside LVGL.

xiasi0, 2025-03-16:
[image]

closed 2025-05-07 by github-actions[bot]