esphome/wake-word-voice-assistants #82: Can't install voice assistant into ATOM Echo

jlatvala, 2025-03-10:
I had a voice assistant working in ATOM Echo but then its entities greyed out in the HA UI. 
Didn't have time or will to start investigating so I decided to install it again from the scratch using the installation procedure in https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/
(note: this same procedure worked fine in the first run **few months ago**)

This time the installation goes through fine but it never ever asks to type in the wifi SSID and password. The next button in the installation UI just doesn't open the SSID/password dialog.
If I try again, it again just re-flashes the SW, but again doesn't ask SSID/password.

Then I tried through this page:
https://esphome.io/projects/ 
Same problem, no luck.

Windows & Chrome.

USB drivers etc. are totally fine, because I can install "dummy" (but working) configuration into ATOM Echo using HA's ESPHome Device Builder add-on and web.esphome.io. This "dummy" configuration can be connected into wifi and it becomes visible into HA. I can also reflash it again using wifi, so the wifi component in ATOM Echo works. 

I also tried to install this yaml
https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml
manually using HA's ESPHome Device Builder add-on and web.esphome.io.
It installs fine, but I never ever get it to work or connected to wifi. 
Tried to modify the yaml also a little, for example removing name_add_mac_suffix: true, using on_connect - delay 5s in wifi-configuration, with and without ap, ...

Sometimes the ATOM Echo makes some kind of continous ticking noise after plugging it into powers (maybe "trying to connect and failing" sound?) 

When trying to use this github's m5stack-atom-echo.yaml, I tried to use the very same !secret wifi_ssid & password, which works in the above-mentioned "dummy" configuration, so the problem should not be the invalid SSID/password.

Tried about 20 times and the gave up and ended up writing this ticket.

Mariusthvdb, 2025-03-18:
yes, I went through all of that the last 2 days....

I finally managed to refresh it via readymade projects attached to the computer, and after that, with some manual editing in the config flow, adopted the device into the Esphome dashboard.

Mind you, an edit and install now takes 30 minutes or so.... it's way beyond ridiculous. 

imho, the firmware has one way and beyond the devices reliable possibilities, and should be made smaller, not including all new developments.

It worked ok-ish before, but ever since the move to voice assistants, has been nothing but trouble.

like you, it updates allright, but seems to nt connect to wifi, or complain about encryption modes, all of which just worked before.

Mariusthvdb, 2025-03-18:
@jlatvala did you check the name of the actual device, and the mentioned named in the log?
notice, like in my logging there is no extension after the `m5stack-atom-echo` part in the `m5stack-atom-echo.local`
while the device is called m5stack-atom-echo-a82d10 in my config... making the MDNS fluke on that.

If you reflash the device, at some point during the addition to Home Assistant, the flow will ask for the name of the device, make sure that is the full name.

Id fixed the connection for me, with help from @BigBobbas on Discord, much appreciated.

jlatvala, 2025-03-28:
I spent some time playing with ESPHome Devide Builder and from a quick look it looks like that the media_player section in 
https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml
breaks the config. Didn't spend too much on it, so the root cause might be somewhere else.
Atom Echo can connect to my wifi and is thus visible in HA without media_player section 
(of course I had to tweak the config in other areas as well to be able to remove media_player)

jlatvala, 2025-04-30:
Tried again yesterday, still not working

jesserockz, 2025-05-22:
Please try this again with the [latest changes](https://github.com/esphome/wake-word-voice-assistants/pull/99) that were merged in, or re-install your echo from https://esphome.io/projects/ after I make a release on this repo very soon.

closed 2025-05-22 by jesserockz

jlatvala, 2025-05-23:
Unfortunately esphome.io/projects or HA ESPHome Device Builder + web.esphome.io didn't work. Flashing works but after that it never asks Wifi details (if using esphome.io/projects) or joins wifi (if using HA ESPHome Device Builder + web.esphome.io). 

Using "Configure WiFi" in web.esphome.io, an error is shown:
"An error occurred. Improv Wi-Fi Serial not detected"

Flashing an "empty" skeleton in ESPHome Device Builder works. Device connects to WiFi and can be added into HA.

jesserockz, 2025-05-26:
@jlatvala Can you provide the logs from the browser console when trying to install from esphome.io/projects?

jlatvala, 2025-05-26:
> [@jlatvala](https://github.com/jlatvala) Can you provide the logs from the browser console when trying to install from esphome.io/projects?

Thanks for trying to fix this.

https://esphome.io/projects/
- Connect button
- Install m5stack.atom-echo-wake-word-voice-assistant button
- Instllation complete UI
- Next button
- (does not ask WiFi SSID & password)
- Shows buttons for 'Install m5stack.atom...' and 'Logs & Console'
- Logs & Console button
- Logs:
`[07:22:47]ets Jun  8 2016 00:22:57
[07:22:47]
[07:22:47]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[07:22:47]configsip: 188777542, SPIWP:0xee
[07:22:47]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[07:22:47]mode:DIO, clock div:2
[07:22:47]load:0x3fff0030,len:7112
[07:22:47]load:0x40078000,len:15688
[07:22:47]load:0x40080400,len:4
[07:22:47]load:0x40080404,len:3876
[07:22:47]entry 0x4008064c
[07:22:47]I (30) boot: ESP-IDF 5.1.6 2nd stage bootloader
[07:22:47]I (30) boot: compile time May 22 2025 23:58:28
[07:22:47]I (30) boot: Multicore bootloader
[07:22:47]I (34) boot: chip revision: v1.1
[07:22:47]I (38) boot.esp32: SPI Speed      : 40MHz
[07:22:47]I (42) boot.esp32: SPI Mode       : DIO
[07:22:47]I (47) boot.esp32: SPI Flash Size : 4MB
[07:22:47]I (51) boot: Enabling RNG early entropy source...
[07:22:47]I (57) boot: Partition Table:
[07:22:47]I (60) boot: ## Label            Usage          Type ST Offset   Length
[07:22:47]I (68) boot:  0 otadata          OTA data         01 00 00009000 00002000
[07:22:47]I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[07:22:47]I (83) boot:  2 app0             OTA app          00 10 00010000 001c0000
[07:22:47]I (90) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[07:22:47]I (98) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[07:22:47]I (105) boot: End of partition table
[07:22:47]I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=79d0ch (498956) map
[07:22:47]I (298) esp_image: segment 1: paddr=00089d34 vaddr=3ffb0000 size=03c14h ( 15380) load
[07:22:47]I (304) esp_image: segment 2: paddr=0008d950 vaddr=40080000 size=026c8h (  9928) load
[07:22:47]I (309) esp_image: segment 3: paddr=00090020 vaddr=400d0020 size=ff8bch (1046716) map
[07:22:47]I (690) esp_image: segment 4: paddr=0018f8e4 vaddr=400826c8 size=1715ch ( 94556) load
[07:22:47]I (742) boot: Loaded app from partition at offset 0x10000
[07:22:47]I (743) boot: Disabling RNG early entropy source...
[07:22:47]I (754) cpu_start: Multicore app
[07:22:47]I (755) quad_psram: This chip is ESP32-PICO
[07:22:47]ets Jun  8 2016 00:22:57
[07:22:47]
[07:22:47]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[07:22:47]configsip: 188777542, SPIWP:0xee
[07:22:47]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[07:22:47]mode:DIO, clock div:2
[07:22:47]load:0x3fff0030,len:7112
[07:22:47]load:0x40078000,len:15688
[07:22:47]load:0x40080400,len:4
[07:22:47]load:0x40080404,len:3876
[07:22:47]entry 0x4008064c
[07:22:47]I (30) boot: ESP-IDF 5.1.6 2nd stage bootloader
[07:22:47]I (30) boot: compile time May 22 2025 23:58:28
[07:22:47]I (30) boot: Multicore bootloader
[07:22:47]I (34) boot: chip revision: v1.1
[07:22:47]I (38) boot.esp32: SPI Speed      : 40MHz
[07:22:47]I (42) boot.esp32: SPI Mode       : DIO
[07:22:47]I (47) boot.esp32: SPI Flash Size : 4MB
[07:22:47]I (51) boot: Enabling RNG early entropy source...
[07:22:47]I (57) boot: Partition Table:
[07:22:47]I (60) boot: ## Label            Usage          Type ST Offset   Length
[... 74 lines cut ...]
[07:23:05]I (42) boot.esp32: SPI Mode       : DIO
[07:23:05]I (47) boot.esp32: SPI Flash Size : 4MB
[07:23:05]I (51) boot: Enabling RNG early entropy source...
[07:23:05]I (57) boot: Partition Table:
[07:23:05]I (60) boot: ## Label            Usage          Type ST Offset   Length
[07:23:05]I (68) boot:  0 otadata          OTA data         01 00 00009000 00002000
[07:23:05]I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[07:23:05]I (83) boot:  2 app0             OTA app          00 10 00010000 001c0000
[07:23:05]I (90) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[07:23:05]I (98) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[07:23:05]I (105) boot: End of partition table
[07:23:05]I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=79d0ch (498956) map
[07:23:05]I (298) esp_image: segment 1: paddr=00089d34 vaddr=3ffb0000 size=03c14h ( 15380) load
[07:23:05]I (304) esp_image: segment 2: paddr=0008d950 vaddr=40080000 size=026c8h (  9928) load
[07:23:05]I (309) esp_image: segment 3: paddr=00090020 vaddr=400d0020 size=ff8bch (1046716) map
[07:23:05]I (690) esp_image: segment 4: paddr=0018f8e4 vaddr=400826c8 size=1715ch ( 94556) load
[07:23:05]I (742) boot: Loaded app from partition at offset 0x10000
[07:23:05]I (742) boot: Disabling RNG early entropy source...
[07:23:05]I (754) cpu_start: Multicore app
[07:23:05]I (755) quad_psram: This chip is ESP32-PICO`

Then:
https://web.esphome.io/
- Connect button
- Menu - Configure Wi-Fi: An error occurred. Improv Wi-Fi Serial not detected

jesserockz, 2025-05-27:
I fear something is physically/electrically wrong with your atom echo. I have at least 2 atom echo's here that install and run just fine from the project page.
Yours looks to be bootlooping due to the watchdog resetting the cpu which is not a good sign.

jlatvala, 2025-05-27:
@jesserockz So you see the WiFi SSID/password dialog after flashing and using the next button?

My echo might be broken even it works when I flash "empty" FW. I can then add it into WiFi and see it in HA.