esphome/wake-word-voice-assistants #88: m5stack echo media player stack overflow

DavidWAbrahams, 2025-03-21:
The main symptom is that my voice assistant works processes commands (lights switch on and off) but I never get a verbal response. In logs, I can see that the m5stack echo is hitting a stack overflow while trying to play the response message.

I can also reproduce the issue by going to the Media Player entity in Home Assistant and having it play even a tiny WAV file from Home Assistant text-to-speech

```
[16:14:25][D][media_player:073]: 'M5Stack Atom Echo 0abbc4' - Setting
[D][media_player:080]:   Media URL: https://<MY SERVER ADDRESS>:38327/api/esphome/ffmpeg_proxy/10145d61f69fb10b63603c813827391b/r-fdol3IPhJt0ANopns3kw.wav
[D][media_player:086]:  Announcement: yes
[D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[D][light:036]: 'M5Stack Atom Echo 0abbc4' Setting:
[D][light:051]:   Brightness: 60%
[D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[D][light:036]: 'M5Stack Atom Echo 0abbc4' Setting:
[D][light:051]:   Brightness: 100%
[D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[D][speaker_media_player:420]: State changed to ANNOUNCING
[D][micro_wake_word:129]: Stopping Microphone
[D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[D][esp-idf:000]: I (555797) I2S: DMA queue destroyed

[D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE

[16:14:25]***ERROR*** A stack overflow in task ann_read has been detected.
[16:14:25]
[16:14:25]
[16:14:25]Backtrace: 0x400833ed:0x3fff4ba0 0x4008ce89:0x3fff4bc0 0x4009022d:0x3fff4be0 0x4008ebb5:0x3fff4c60 0x4008cf84:0x3fff4c90 0x4008cf36:0x4000bff0 |<-CORRUPTED
[16:14:25]
[16:14:25]
[16:14:25]
[16:14:25]
[16:14:25]ELF file SHA256: 42fea1c496829731
[16:14:25]
[16:14:25]Rebooting...
[16:14:25]ets Jun  8 2016 00:22:57
[16:14:25]
[16:14:25]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:14:25]configsip: 188777542, SPIWP:0xee
[16:14:25]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:14:25]mode:DIO, clock div:2
[16:14:25]load:0x3fff0030,len:6652
[16:14:25]ho 0 tail 12 room 4
[16:14:25]load:0x40078000,len:15052
[16:14:25]load:0x40080400,len:3836
[16:14:25]entry 0x4008069c
```

Things that might be odd about my setup:
* I use a HTTP URL for my home assistant server. 
* I have an error about missing PSRAM in the m5stack echo log during boot. I don't know if this is a hardware issue or something else:
```
[16:14:25]I (703) psram: This chip is ESP32-PICO
[16:14:25]E (703) psram: PSRAM ID read error: 0xffffffff
[16:14:25]E (705) spiram: SPI RAM enabled but initialization failed. Bailing out.
```

DavidWAbrahams, 2025-03-22:
@kahrendt and @kbx81 since this is tied to your [speaker to media player migration](https://github.com/esphome/wake-word-voice-assistants/commit/389be0bf1dd5e45594ac645aa4cf823c771c04b3)

DavidWAbrahams, 2025-03-24:
BTW this is probably a duplicate of https://github.com/esphome/wake-word-voice-assistants/issues/71
(the media player announce feature breaks on HTTPS connections)
I temporarily worked around the issue by adding this to my yaml:
`
http_request:
  verify_ssl: false
`

djtecha, 2025-03-27:
Are you using this with the echo package or creating the entire yaml? Would this work from ESPHome Builder in HA?

```
substitutions:
  name: atom-echo-3
  friendly_name: atom-echo-3
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
api:
  encryption:
    key: REDACT
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

closed 2025-05-07 by github-actions[bot]