esphome/wake-word-voice-assistants #98: M5Stack Atom Echo compile fails with esphome 2025.4.0 - esp_psram.h: No such file or directory

darth-hp, 2025-04-17:
Hey there,

Compile fails with the latest upgrade:

```
|-- esp-audio-libs @ 1.1.3
Compiling .pioenvs/m5stack-atom-echo-0ac384/src/esphome/components/psram/psram.o
Compiling .pioenvs/m5stack-atom-echo-0ac384/src/esphome/components/safe_mode/safe_mode.o
Compiling .pioenvs/m5stack-atom-echo-0ac384/src/esphome/components/script/script.o
Compiling .pioenvs/m5stack-atom-echo-0ac384/src/esphome/components/select/select.o
src/esphome/components/psram/psram.cpp:5:10: fatal error: esp_psram.h: No such file or directory
```

YAML config:

``` yaml
substitutions:
  name: m5stack-atom-echo-012345
  friendly_name: Atom-01
packages:
  m5stack.atom-echo-wake-word-voice-assistant: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: jGo1Nokia1qzp1NfWj1uvHm1aW1OkLionXN11ynx1Yw=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

voyagerft, 2025-04-17:
same problem

spudje, 2025-04-17:
It won't compile for me either, I get this:
Home Assistant 2025.4.1
ESPHome

`Compiling .pioenvs/m5stack-atom-echo-0d03dc/components/esp-tflite-micro/tensorflow/lite/micro/micro_allocator.o
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/m5stack-atom-echo-0d03dc/src/main.o] Error 1`

bilbotbaggin, 2025-04-17:
I can confirm the same problem on my two echos, both using the stock wake word voice assistant configuration.

benson789, 2025-04-17:
Same problem here on two different echos.

Marcello2022, 2025-04-17:
Same here but with help of Claude AI, I added the lines below and the compliation ran successfully. Not sure though whether this is the "right" fix, for this we need feedback from the devs, I guess...

`substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: alexa  # okay_nabu, hey_jarvis, hey_mycroft are also supported

external_components:
  - source: github://esphome/esphome@2024.4.0
    components: [psram]

esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
  min_version: 2025.2.0`

darth-hp, 2025-04-17:
You could also just use esphome 2025.3.3 probably again 😁

```
 external_components:
 - source: github://esphome/esphome@2025.3.3
   components: [psram]
```

Spamfast, 2025-04-17:
> You could also just use esphome 2025.3.3 probably again 😁

Good point. Just checked - and yes, compiles with that too. Probably a better option to be 'nearer' as well.

(Obviously they can't check all our stuff but one might think that the ESPHome CI server might attempt compile of all github://esphome/.... 'main' & latest release packages before releasing a new tag, and flag those that no longer work. Hey ho.)

RoGueMS, 2025-04-17:
> Good point. Just checked - and yes, compiles with that too. Probably a better option to be 'nearer' as well.
>
> (Obviously they can't check all our stuff but one might think that the ESPHome CI server [...]

Is it possible for you to share your YAML file up until the **esphome:** field?

RoGueMS, 2025-04-17:
@Spamfast thank you, i did that and it's compiling atm waiting to be able to download and then flash! 👍🏻

Spamfast, 2025-04-17:
> [@Spamfast](https://github.com/Spamfast) thank you, i did that and it's compiling atm waiting to be able to download and then flash! 👍🏻

My pleasure - although it was @Marcello2022 who posted the fix & @darth-hp who suggested pulling from the more recent tag. Just remember to remove the external component entry once you're using whatever release of ESPHome integrates the 'proper' fix.

RoGueMS, 2025-04-17:
> My pleasure - although it was [@Marcello2022](https://github.com/Marcello2022) who posted the fix & [@darth-hp](https://github.com/darth-hp) who suggested pulling from the more recent tag. Just [...]

Yes i'll do that, this preparing of the .bin file takes quite a while everytime, it's still compiling, no fatal error yet but a lot of yellow warnings.

charettepa, 2025-04-17:
@Spamfast 

thank you this worked for me
how will we know when its fixed and to remove it

RoGueMS, 2025-04-18:
> [@Spamfast](https://github.com/Spamfast)
> 
> thank you this worked for me how will we know when its fixed and to remove it

from what i could tell the fix will be coming on 2025.4.1 so before that ships i don't think its necessary to change.

MichaelMKKelly, 2025-04-29:
it is indeed fixed in 2025.4.1

Taomyn, 2025-04-29:
Removed the temporary fix and now compiles

darth-hp, 2025-04-29:
Fixed with 2025.4.1

RoGueMS, 2025-04-29:
> Fixed with 2025.4.1

Quick question because this is not my beach 😄 
Do you just remove the lines from the YAML and Save or do you need to recompile and reflash a new firmware?

MichaelMKKelly, 2025-04-29:
> Do you just remove the lines from the YAML and Save or do you need to recompile and reflash a new firmware?

saving only saves the config locally. you have to compile/install it to actually put it on the device