esphome/wake-word-voice-assistants #111: m5stack-atom-echo will not build in ESPHome 2025.5 beta HA add-on

mwhdc, 2025-05-19:
### The problem

Error occurs when building with ESPHome 2025.5 beta packaged as HA add-on. Seems to be related to the older PlatformIO version specified in the device yaml. Manual solution given [here](https://github.com/esphome/issues/issues/5766#issuecomment-2103563641) but making alterations to the add-on package is not desirable.

### Which device are you using?

m5stack-atom-echo

### Firmware Version

Commit fcb1345

### Logs

```txt
INFO ESPHome 2025.5.0b4
INFO Reading configuration /config/esphome/esp-echo1.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.3.1
INFO Compiling app...
Processing esp-echo1 (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
ModuleNotFoundError: No module named 'pkg_resources':
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 142:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 352:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 684:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/frameworks/espidf.py", line 29:
    import pkg_resources
========================== [FAILED] Took 0.52 seconds ==========================
```

voyagerft, 2025-05-21:
Same problem

Taomyn, 2025-05-21:
I got bored and had a go at creating a new yaml for my device, I took the main [yaml](https://raw.githubusercontent.com/esphome/wake-word-voice-assistants/refs/heads/main/m5stack-atom-echo/m5stack-atom-echo.yaml) currently being imported, added the tweaks for my own yaml in HAOS, then amended the parts that wouldn't compile using an example of the [changes](https://github.com/esphome/esphome/pull/8645). So now it compiles, plenty or warnings like before, and flashes successfully.

Be aware, I have no idea if it fully works as I am working away from home, but I do know I can switch the led on/off and this is picked up in the logs and by HAOS.

Here is the yaml:
```
substitutions:
  name: "m5ae-31b338"
  friendly_name: M5Stack Atom Echo 31b338
  micro_wake_word_model: hey_jarvis

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  min_version: 2025.5.0

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    # version: 4.4.8
    # platform_version: 5.4.0

logger:
api:
  encryption:
    key: !secret m5ae_key

ota:
  - platform: esphome
    id: ota_esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.39
    subnet: 255.255.255.0
    gateway: 192.168.1.1
    dns1: 192.168.1.11
    dns2: 192.168.1.71
  domain: !secret dns_suffix
  on_connect:
    - delay: 5s
http_request:
  verify_ssl: false

captive_portal:

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
[... 379 lines cut ...]
[15:50:21][C][micro_wake_word:016]:       Sliding window size: 5
[15:50:21][C][micro_wake_word:020]:     - VAD Model
[15:50:21][C][micro_wake_word:021]:       Probability cutoff: 0.50
[15:50:21][C][micro_wake_word:022]:       Sliding window size: 5
[15:50:23][D][light:036]: 'M5Stack Atom Echo 31b338' Setting:
[15:50:23][D][micro_wake_word:360]: Starting wake word detection
[15:50:23][D][micro_wake_word:378]: State changed from STOPPED to STARTING
[15:50:23][D][ring_buffer:034][mww]: Created ring buffer with size 3840
[15:50:23][D][micro_wake_word:261]: Inference task has started, attempting to allocate memory for buffers
[15:50:23][D][micro_wake_word:266]: Inference task is running
[15:50:23][D][micro_wake_word:378]: State changed from STARTING to DETECTING_WAKE_WORD
[15:50:23][D][i2s_audio.microphone:455]: Task has started, attempting to setup I2S audio driver
[15:50:23][D][i2s_audio.microphone:460]: Task is running and reading data
[15:50:23][W][component:257]: Component i2s_audio.microphone took a long time for an operation (106 ms).
[15:50:23][W][component:258]: Components should block for at most 30 ms.
[15:50:43][D][esp32.preferences:114]: Saving 2 preferences to flash...
[15:50:43][D][esp32.preferences:142]: Saving 2 preferences to flash: 2 cached, 0 written, 0 failed
[15:51:14][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[15:51:14][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:51:14][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:51:14][W][component:257]: Component safe_mode took a long time for an operation (68 ms).
[15:51:14][W][component:258]: Components should block for at most 30 ms.
```

I take no responsibility if this breaks anyone else's device but maybe it can help get this fixed faster when someone more knowledgeable than me gets around to fixing it.

mwhdc, 2025-05-21:
> I got bored and had a go at creating a new yaml for my device, I took the main [...]

Ah good idea! I tried a variation of your approach - I had my local yaml pull the "canonical" Github package and then make a few changes. The firmware built successfully and worked as intended.

```
packages:
  base: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main

substitutions:
  name: atom-echo
  friendly_name: Atom Echo
  micro_wake_word_model: hey_jarvis

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: !remove
    platform_version: !remove

light:
  - platform: esp32_rmt_led_strip
    id: !extend led
    rmt_channel: !remove
```

jesserockz, 2025-05-22:
The YAML in this repo have been [updated now](https://github.com/esphome/wake-word-voice-assistants/pull/99) and build with ESPHome 2025.5.0.
We could not update them before the release as it would mean they were even more broken.

closed 2025-05-22 by jesserockz